nigelb/platform-apollo3blue #5: Compile error

kamejoko80, 2020-10-07:
Hello,

Thanks for providing the framework. I've got the following error when compiling the source code:

src\main.cpp:1:10: fatal error: Arduino.h: No such file or directory

There is no Arduino.h in the source tree.

chepo92, 2020-11-12:
Same issue

tkanarsky, 2020-11-12:
The Sparkfun repo recently transitioned to a backwards-incompatible v2 core that's based on the mbed RTOS. As a result, the arduino.py builder file doesn't properly define the header search paths for the new version of the Arduino framework, hence why you're getting the compile error. 

To fix this, you can manually clone a v1.x.x version of the Arduino_Apollo3 framework and run the steps again. Keep in mind that this version of the framework does not have BLE functionality, as that depends on some low-level magic introduced in the mbed RTOS. @nigelb, have you looked into updating this repo to be compatible with the v2 framework? I'd gladly lend a hand but I'm not familiar with what directories need to be included in the search path.

vinthewrench, 2020-11-16:
OK I am not sure I got the details here,  I copied the directory from 

/Arduino15/packages/SparkFun/hardware/apollo3/1.2.1 to the platformio/packages directory and renamed it framework-arduinoapollo3 

and then tried building a simple test program as your example and when I compile I see the following

platformio run 
Processing SparkFun_RedBoard_Artemis (platform: apollo3blue; board: SparkFun_RedBoard_Artemis; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/apollo3blue/SparkFun_RedBoard_Artemis.html
PLATFORM: Apollo 3 Blue (0.0.1) > SparkFun RedBoard Artemis
HARDWARE: AMA3B1KK 48MHz, 348KB RAM, 937.50KB Flash
PACKAGES: 
 - framework-arduinoapollo3 1.0.23 
 - toolchain-gccarmnoneeabi 1.90301.200702 (9.3.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Linking .pio/build/SparkFun_RedBoard_Artemis/program
/Users/vinnie/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: .pio/build/SparkFun_RedBoard_Artemis/libvariant.a(startup_gcc.c.o): in function `Reset_Handler':
/Users/vinnie/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis/startup/startup_gcc.c:321: undefined reference to `main'
/Users/vinnie/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: .pio/build/SparkFun_RedBoard_Artemis/libapollo3_sdk_utils.a(am_util_delay.c.o): in function `am_util_delay_ms':
/Users/vinnie/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/utils/am_util_delay.c:95: undefined reference to `am_hal_clkgen_status_get'
collect2: error: ld returned 1 exit status
*** [.pio/build/SparkFun_RedBoard_Artemis/program] Error 1
======================================================= [FAILED] Took 1.22 seconds =======================================================

so maybe I misunderstood,  can you give me some details as to what you mean by " manually clone a v1.x.x version of the Arduino_Apollo3 framework and run the steps again."    which steps?
many thanks!

tkanarsky, 2020-11-16:
So these were the steps I took to make it work: 
`cd ~/.platformio/packages/`
`wget https://github.com/sparkfun/Arduino_Apollo3/archive/v1.2.1.zip`
`unzip v1.2.1.zip && mv Arduino_Apollo3-1.2.1 framework-arduinoapollo3 && rm v1.2.1.zip`
`cd ~/.platformio/platforms/`
`git clone https://github.com/nigelb/platform-apollo3blue.git apollo3blue`

After this, go to your project dir and run `pio run`. This was the output when I do it: 

```
Processing SparkFun_Artemis_Nano (platform: apollo3blue; board: SparkFun_Artemis_Nano; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option

Warning! Please install `99-platformio-udev.rules`. 
More details: https://docs.platformio.org/page/faq.html#platformio-udev-rules

Auto-detected: /dev/ttyUSB0
CONFIGURATION: https://docs.platformio.org/page/boards/apollo3blue/SparkFun_Artemis_Nano.html
PLATFORM: Apollo 3 Blue (0.0.1) > SparkFun RedBoard Artemis Nano
HARDWARE: AMA3B1KK 48MHz, 348KB RAM, 937.50KB Flash
PACKAGES: 
 - framework-arduinoapollo3 1.0.23 
 - toolchain-gccarmnoneeabi 1.90301.200702 (9.3.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/SparkFun_Artemis_Nano/src/main.cpp.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/analog/ap3_analog.cpp.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/analog/ap3_analog_structures.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/ard_supers/IPAddress.cpp.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/ard_supers/Print.cpp.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/ard_supers/Stream.cpp.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/ard_supers/WMath.cpp.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/ard_supers/WString.cpp.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/ard_supers/avr/dtostrf.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/ard_supers/hooks.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/ard_supers/itoa.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/clock/ap3_clock_sources.cpp.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/debugging/ap3_debugging.cpp.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/gpio/ap3_gpio.cpp.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/gpio/ap3_gpio_structures.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/gpio/ap3_shift.cpp.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/initialization/ap3_initialization.cpp.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/iomaster/ap3_iomaster.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/iomaster/ap3_iomaster.cpp.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/main.cpp.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/timing/ap3_timing.cpp.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/uart/ap3_uart.cpp.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/uart/ap3_uart_structures.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_adc.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_ble.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_ble_patch.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_ble_patch_b0.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_burst.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_cachectrl.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_clkgen.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_cmdq.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_ctimer.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_debug.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_flash.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_global.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_gpio.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_interrupt.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_iom.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_ios.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_itm.c.o
[... 51 lines cut ...]
Indexing .pio/build/SparkFun_Artemis_Nano/libServo.a
Archiving .pio/build/SparkFun_Artemis_Nano/libRTC.a
Archiving .pio/build/SparkFun_Artemis_Nano/libPDM.a
Indexing .pio/build/SparkFun_Artemis_Nano/libRTC.a
Indexing .pio/build/SparkFun_Artemis_Nano/libPDM.a
Archiving .pio/build/SparkFun_Artemis_Nano/libSPI.a
Indexing .pio/build/SparkFun_Artemis_Nano/libSPI.a
Archiving .pio/build/SparkFun_Artemis_Nano/libSoftwareSerial.a
Indexing .pio/build/SparkFun_Artemis_Nano/libSoftwareSerial.a
Archiving .pio/build/SparkFun_Artemis_Nano/libWire.a
Indexing .pio/build/SparkFun_Artemis_Nano/libWire.a
Linking .pio/build/SparkFun_Artemis_Nano/program
[... cut ...]
Flash: [          ]   0.1% (used 988 bytes from 960000 bytes)
========================= [SUCCESS] Took 2.37 seconds =========================

********************************************************************************
There is a new version 5.0.3 of PlatformIO available.
Please upgrade it via `platformio upgrade` or `pip install -U platformio` command.
Changes: https://docs.platformio.org/en/latest/history.html
********************************************************************************
```

vinthewrench, 2020-11-16:
thank you.. thats what I thought --  the difference is when I linked:

Linking .pio/build/SparkFun_RedBoard_Artemis/program
/Users/vinnie/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: .pio/build/SparkFun_RedBoard_Artemis/libvariant.a(startup_gcc.c.o): in function `Reset_Handler':
/Users/vinnie/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis/startup/startup_gcc.c:321: undefined reference to `main'
/Users/vinnie/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: .pio/build/SparkFun_RedBoard_Artemis/libapollo3_sdk_utils.a(am_util_delay.c.o): in function `am_util_delay_ms':
/Users/vinnie/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/utils/am_util_delay.c:95: undefined reference to `am_hal_clkgen_status_get'
collect2: error: ld returned 1 exit status
*** [.pio/build/SparkFun_RedBoard_Artemis/program] Error 1
================================================= [FAILED] Took 3.82 seconds =================================================

I wonder if there is a caching issue?  is there a cache folder I need to clear?

vinthewrench, 2020-11-16:
oh I am macOS?  maybe that is an issue?

tkanarsky, 2020-11-16:
Try running `platformio run -t clean` and then `platformio run`

vinthewrench, 2020-11-16:
yes did that first -- I also updated it to a nano to be sure we were on same page -- same result 

vinnie@ZeroDarkPro test % platformio run -t clean
Processing SparkFun_Artemis_Nano (platform: apollo3blue; board: SparkFun_Artemis_Nano; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------
Removed .pio/build/SparkFun_Artemis_Nano/libPDM.a
Removed .pio/build/SparkFun_Artemis_Nano/program.map
Removed .pio/build/SparkFun_Artemis_Nano/libSoftwareSerial.a
Removed .pio/build/SparkFun_Artemis_Nano/libapollo3_sdk_devices.a
Removed .pio/build/SparkFun_Artemis_Nano/libEEPROM.a
Removed .pio/build/SparkFun_Artemis_Nano/libWire.a
Removed .pio/build/SparkFun_Artemis_Nano/.sconsign38.dblite
Removed .pio/build/SparkFun_Artemis_Nano/libRTC.a
Removed .pio/build/SparkFun_Artemis_Nano/libSPI.a
Removed .pio/build/SparkFun_Artemis_Nano/libapollo3_sdk_mcu.a
Removed .pio/build/SparkFun_Artemis_Nano/libvariant.a
Removed .pio/build/SparkFun_Artemis_Nano/libFrameworkArduino.a
Removed .pio/build/SparkFun_Artemis_Nano/libapollo3_sdk_utils.a
Removed .pio/build/SparkFun_Artemis_Nano/libServo.a
Removed .pio/build/SparkFun_Artemis_Nano/Wire/Wire.cpp.d
Removed .pio/build/SparkFun_Artemis_Nano/Wire/Wire.cpp.o
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_flash.c.o
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_rtc.c.o
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_secure_ota.c.o
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_mcuctrl.c.o
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_ctimer.c.o
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_ble.c.d
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_security.c.d
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_ble_patch.c.d
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_ble_patch_b0.c.d
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_global.c.d
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_iom.c.o
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_burst.c.o
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_pwrctrl.c.o
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_systick.c.o
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_wdt.c.d
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_gpio.c.o
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_interrupt.c.o
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_pdm.c.o
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_cmdq.c.d
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_scard.c.o
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_queue.c.o
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_ios.c.d
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_tpiu.c.o
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_adc.c.o
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_stimer.c.d
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_uart.c.d
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_debug.c.o
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_sysctrl.c.o
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_reset.c.o
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_itm.c.d
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_mspi.c.o
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_cachectrl.c.o
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_clkgen.c.d
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_ble.c.o
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_ctimer.c.d
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_secure_ota.c.d
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_rtc.c.d
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_flash.c.d
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_mcuctrl.c.d
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_iom.c.d
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_ble_patch_b0.c.o
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_ble_patch.c.o
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_security.c.o
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_global.c.o
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_wdt.c.o
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_pdm.c.d
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_interrupt.c.d
Removed .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_gpio.c.d
[... cut ...]
/Users/vinnie/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: .pio/build/SparkFun_Artemis_Nano/libapollo3_sdk_utils.a(am_util_delay.c.o): in function `am_util_delay_ms':
/Users/vinnie/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/utils/am_util_delay.c:95: undefined reference to `am_hal_clkgen_status_get'
collect2: error: ld returned 1 exit status
*** [.pio/build/SparkFun_Artemis_Nano/program] Error 1
================================================= [FAILED] Took 3.87 seconds =================================================
vinnie@ZeroDarkPro test %

vinthewrench, 2020-11-17:
except for the order or indexing and archiving our compiles are the same.. what I fail i is the linking

tkanarsky, 2020-11-17:
Can you open a separate issue? This issue relates to the mbed core incompatibility; what you're experiencing is something else. Thanks!

vinthewrench, 2020-11-17:
no problem -- https://github.com/nigelb/platform-apollo3blue/issues/6

nigelb, 2020-11-20:
Hi everyone, I will have a look into upgrading to the v2 framework. Hopefully it is just a matter of fixing the paths.

jerabaul29, 2020-12-05:
Would it be possible to update the installation instructions in the readme so that they work given the current states of the different repos? :) That would be very helpful :)

jerabaul29, 2020-12-11:
Any update regarding the upgrading to v2 framework? :) Would be really useful to be able to use the mbed-os core :)

closed 2021-06-10 by nigelb