nigelb/platform-apollo3blue #6: Linking error -  v1.x.x version of the Arduino_Apollo3 framework

vinthewrench, 2020-11-17:
hello;

On macOS 10.15
I have followed the directions for installing platform-apollo3blue using the 1.0.23 framework,.
(I understand the issues with the 2,.0 library )

--
cd ~/.platformio/packages/
wget https://github.com/sparkfun/Arduino_Apollo3/archive/v1.2.1.zip
unzip v1.2.1.zip && mv Arduino_Apollo3-1.2.1 framework-arduinoapollo3 && rm v1.2.1.zip
cd ~/.platformio/platforms/
git clone https://github.com/nigelb/platform-apollo3blue.git apollo3blue
--

and created the simple SparkFun_Artemis_Nano app ,

I make it through the compile, but fail in the link

pio run                
Processing SparkFun_Artemis_Nano (platform: apollo3blue; board: SparkFun_Artemis_Nano; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Auto-detected: /dev/cu.usbserial-1440
CONFIGURATION: https://docs.platformio.org/page/boards/apollo3blue/SparkFun_Artemis_Nano.html
PLATFORM: Apollo 3 Blue (0.0.1) > SparkFun RedBoard Artemis Nano
HARDWARE: AMA3B1KK 48MHz, 348KB RAM, 937.50KB Flash
PACKAGES: 
 - framework-arduinoapollo3 1.0.23 
 - toolchain-gccarmnoneeabi 1.90301.200702 (9.3.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/SparkFun_Artemis_Nano/src/SparkFun_Artemis_Nano.cpp.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/analog/ap3_analog.cpp.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/analog/ap3_analog_structures.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/ard_supers/IPAddress.cpp.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/ard_supers/Print.cpp.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/ard_supers/Stream.cpp.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/ard_supers/WMath.cpp.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/ard_supers/WString.cpp.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/ard_supers/avr/dtostrf.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/ard_supers/hooks.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/ard_supers/itoa.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/clock/ap3_clock_sources.cpp.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/debugging/ap3_debugging.cpp.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/gpio/ap3_gpio.cpp.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/gpio/ap3_gpio_structures.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/gpio/ap3_shift.cpp.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/initialization/ap3_initialization.cpp.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/iomaster/ap3_iomaster.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/iomaster/ap3_iomaster.cpp.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/main.cpp.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/timing/ap3_timing.cpp.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/uart/ap3_uart.cpp.o
Compiling .pio/build/SparkFun_Artemis_Nano/FrameworkArduino/uart/ap3_uart_structures.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_adc.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_ble.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_ble_patch.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_ble_patch_b0.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_burst.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_cachectrl.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_clkgen.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_cmdq.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_ctimer.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_debug.c.o
Compiling .pio/build/SparkFun_Artemis_Nano/apollo3_sdk_mcu/apollo3/hal/am_hal_flash.c.o
Archiving .pio/build/SparkFun_Artemis_Nano/libFrameworkArduino.a
[... 50 lines cut ...]
Compiling .pio/build/SparkFun_Artemis_Nano/Wire/Wire.cpp.o
Archiving .pio/build/SparkFun_Artemis_Nano/libvariant.a
Indexing .pio/build/SparkFun_Artemis_Nano/libvariant.a
Archiving .pio/build/SparkFun_Artemis_Nano/libEEPROM.a
Indexing .pio/build/SparkFun_Artemis_Nano/libEEPROM.a
Archiving .pio/build/SparkFun_Artemis_Nano/libServo.a
Indexing .pio/build/SparkFun_Artemis_Nano/libServo.a
Archiving .pio/build/SparkFun_Artemis_Nano/libPDM.a
Archiving .pio/build/SparkFun_Artemis_Nano/libRTC.a
Indexing .pio/build/SparkFun_Artemis_Nano/libPDM.a
Indexing .pio/build/SparkFun_Artemis_Nano/libRTC.a
Archiving .pio/build/SparkFun_Artemis_Nano/libSPI.a
Indexing .pio/build/SparkFun_Artemis_Nano/libSPI.a
Archiving .pio/build/SparkFun_Artemis_Nano/libWire.a
Indexing .pio/build/SparkFun_Artemis_Nano/libWire.a
Archiving .pio/build/SparkFun_Artemis_Nano/libSoftwareSerial.a
Indexing .pio/build/SparkFun_Artemis_Nano/libSoftwareSerial.a
Linking .pio/build/SparkFun_Artemis_Nano/program
/Users/vinnie/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: .pio/build/SparkFun_Artemis_Nano/libvariant.a(startup_gcc.c.o): in function `Reset_Handler':
/Users/vinnie/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis_nano/startup/startup_gcc.c:321: undefined reference to `main'
/Users/vinnie/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: .pio/build/SparkFun_Artemis_Nano/libapollo3_sdk_utils.a(am_util_delay.c.o): in function `am_util_delay_ms':
/Users/vinnie/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/utils/am_util_delay.c:95: undefined reference to `am_hal_clkgen_status_get'
collect2: error: ld returned 1 exit status
*** [.pio/build/SparkFun_Artemis_Nano/program] Error 1
=============================================== [FAILED] Took 4.14 seconds ===============================================

vinthewrench, 2020-11-17:
verbose 

arm-none-eabi-g++ -o .pio/build/SparkFun_Artemis_Nano/variant/config/variant.cpp.o -c -std=gnu++11 -std=gnu++11 -fno-threadsafe-statics -fno-rtti -c -g -MMD -mthumb -mfloat-abi=hard -fdata-sections -Os -ffunction-sections -nostdlib --param max-inline-insns-single=500 -fno-exceptions -mcpu=cortex-m4 -DPART_apollo3 -DAM_PACKAGE_BGA -DAM_PART_APOLLO3 -DARDUINO_ARCH_APOLLO3 -DPLATFORMIO=50003 -DARDUINO_AM_AP3_SFE_BB_ARTEMIS_NANO -DF_CPU=48000000L -DARDUINO=10809 -I/Users/vinnie/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis_nano/config -I/Users/vinnie/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis_nano/bsp -I/Users/vinnie/.platformio/packages/framework-arduinoapollo3/cores/arduino/ard_sup -I/Users/vinnie/.platformio/packages/framework-arduinoapollo3/cores/arduino/ard_sup/ard_supers -I/Users/vinnie/.platformio/packages/framework-arduinoapollo3/cores/arduino -I/Users/vinnie/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis_nano -I/Users/vinnie/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/mcu/apollo3 -I/Users/vinnie/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/mcu/apollo3/hal -I/Users/vinnie/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/mcu/apollo3/regs -I/Users/vinnie/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/utils -I/Users/vinnie/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/devices -I/Users/vinnie/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/CMSIS/AmbiqMicro/Include -I/Users/vinnie/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/CMSIS/ARM/Include -I/Users/vinnie/.platformio/packag/Users/vinnie/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: .pio/build/SparkFun_Artemis_Nano/libvariant.a(startup_gcc.c.o): in function `Reset_Handler':
/Users/vinnie/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis_nano/startup/startup_gcc.c:321: undefined reference to `main'
/Users/vinnie/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: .pio/build/SparkFun_Artemis_Nano/libapollo3_sdk_utils.a(am_util_delay.c.o): in function `am_util_delay_ms':
/Users/vinnie/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/utils/am_util_delay.c:95: undefined reference to `am_hal_clkgen_status_get'
collect2: error: ld returned 1 exit status

nigelb, 2020-11-20:
Hi @vinthewrench, I found the same issue on Linux. It would seem that the problem stems from a bug introduced in PlatformIO (see: platformio/platformio-core#3674). I only found out what the problem is because on one of my PCs I have an old version of PlatformIO installed and it did not exhibit the issue and I could compare the liker command:

> arm-none-eabi-g++ -o .pio/build/SparkFun_Artemis_Nano/program -T/home/nigelb/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis_nano/linker_scripts/gcc/artemis_sbl_svl_app.ld -mthumb -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard --specs=nano.specs -mfloat-abi=hard -nostdlib -fno-exceptions -static -Wl,--gc-sections,--entry,Reset_Handler -Wl,--check-sections -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -Wl,-Map=/home/nigelb/projects/ucontroller/Apollo3_PIO/.pio/build/SparkFun_Artemis_Nano/program.map .pio/build/SparkFun_Artemis_Nano/src/main.cpp.o -L.pio/build/SparkFun_Artemis_Nano -L/home/nigelb/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis_nano -L/home/nigelb/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/CMSIS/ARM/Lib/ARM .pio/build/SparkFun_Artemis_Nano/libFrameworkArduino.a .pio/build/SparkFun_Artemis_Nano/libapollo3_sdk_mcu.a .pio/build/SparkFun_Artemis_Nano/libapollo3_sdk_devices.a .pio/build/SparkFun_Artemis_Nano/libapollo3_sdk_utils.a .pio/build/SparkFun_Artemis_Nano/libvariant.a .pio/build/SparkFun_Artemis_Nano/libEEPROM.a .pio/build/SparkFun_Artemis_Nano/libPDM.a .pio/build/SparkFun_Artemis_Nano/libRTC.a .pio/build/SparkFun_Artemis_Nano/libServo.a .pio/build/SparkFun_Artemis_Nano/libSoftwareSerial.a .pio/build/SparkFun_Artemis_Nano/libSPI.a .pio/build/SparkFun_Artemis_Nano/libWire.a -lm -larm_cortexM4lf_math -lgcc -lstdc++ -lnosys -lc

vs

> arm-none-eabi-g++ -o .pio/build/SparkFun_Artemis_Nano/program -T/home/nigelb/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis_nano/linker_scripts/gcc/artemis_sbl_svl_app.ld -mthumb -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard --specs=nano.specs -mfloat-abi=hard -nostdlib -fno-exceptions -static -Wl,--gc-sections,--entry,Reset_Handler -Wl,--check-sections -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -Wl,-Map=/home/nigelb/projects/ucontroller/Apollo3_PIO/.pio/build/SparkFun_Artemis_Nano/program.map .pio/build/SparkFun_Artemis_Nano/src/main.cpp.o -L.pio/build/SparkFun_Artemis_Nano -L/home/nigelb/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis_nano -L/home/nigelb/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/CMSIS/ARM/Lib/ARM -Wl,--start-group .pio/build/SparkFun_Artemis_Nano/libFrameworkArduino.a .pio/build/SparkFun_Artemis_Nano/libapollo3_sdk_mcu.a .pio/build/SparkFun_Artemis_Nano/libapollo3_sdk_devices.a .pio/build/SparkFun_Artemis_Nano/libapollo3_sdk_utils.a .pio/build/SparkFun_Artemis_Nano/libvariant.a .pio/build/SparkFun_Artemis_Nano/libEEPROM.a .pio/build/SparkFun_Artemis_Nano/libPDM.a .pio/build/SparkFun_Artemis_Nano/libRTC.a .pio/build/SparkFun_Artemis_Nano/libServo.a .pio/build/SparkFun_Artemis_Nano/libSoftwareSerial.a .pio/build/SparkFun_Artemis_Nano/libSPI.a .pio/build/SparkFun_Artemis_Nano/libWire.a -lm -larm_cortexM4lf_math -lgcc -lstdc++ -lnosys -lc -Wl,--end-group

The problem being the missing ` -Wl,--start-group` and `-Wl,--end-group`, anyway the fix recomended in platformio/platformio-core#3674 in is `pio upgrade --dev`. I tested it and the linker links fine once I upgraded to:

> pio --version
>PlatformIO Core, version 5.0.4a2

tsandmann, 2021-03-23:
FYI: see https://github.com/Wiz-IO/wizio-pico/issues/24#issuecomment-805256842 for the root cause of the problem (on macOS) and a possible solution/workaround.

closed 2021-06-10 by nigelb