nigelb/platform-apollo3blue #7: upload_port issue when trying to compile

jerabaul29, 2020-12-05:
I encounter some issues when trying to compile, any idea how to fix that?

```
~/Desktop/Current/test_artemis$ platformio run -v
Processing SparkFun_RedBoard_Artemis (platform: apollo3blue; board: SparkFun_RedBoard_Artemis; framework: arduino)
------------------------------------------------------------------------------------------------------------------------

Warning! Please install `99-platformio-udev.rules`. 
More details: https://docs.platformio.org/page/faq.html#platformio-udev-rules

Error: Please specify `upload_port` for environment or use global `--upload-port` option.
For some development platforms it can be a USB flash drive (i.e. /media/<user>/<device name>)
============================================== [FAILED] Took 0.24 seconds ==============================================
```

jerabaul29, 2020-12-05:
Ok, I can specify the port by hand, but then I hit another problem:

```
~/Desktop/Current/test_artemis$ platformio run -v --upload-port /dev/ttyUSB0
Processing SparkFun_RedBoard_Artemis (platform: apollo3blue; board: SparkFun_RedBoard_Artemis; framework: arduino)
------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/apollo3blue/SparkFun_RedBoard_Artemis.html
PLATFORM: Apollo 3 Blue (0.0.1) > SparkFun RedBoard Artemis
HARDWARE: AMA3B1KK 48MHz, 348KB RAM, 937.50KB Flash
PACKAGES: 
 - framework-arduinoapollo3 1.0.22 
 - toolchain-gccarmnoneeabi 1.90301.200702 (9.3.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
arm-none-eabi-g++ -o .pio/build/SparkFun_RedBoard_Artemis/src/main.cpp.o -c -std=gnu++11 -std=gnu++11 -fno-threadsafe-statics -fno-rtti -c -g -MMD -mthumb -mfloat-abi=hard -fdata-sections -Os -ffunction-sections -nostdlib --param max-inline-insns-single=500 -fno-exceptions -mcpu=cortex-m4 -DPART_apollo3 -DAM_PACKAGE_BGA -DAM_PART_APOLLO3 -DARDUINO_ARCH_APOLLO3 -DPLATFORMIO=50003 -DARDUINO_AM_AP3_SFE_BB_ARTEMIS -DF_CPU=48000000L -DARDUINO=10809 -Iinclude -Isrc -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis/config -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis/bsp -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/ard_sup -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/ard_sup/ard_supers -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/mcu/apollo3 -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/mcu/apollo3/hal -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/mcu/apollo3/regs -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/utils -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/devices -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/CMSIS/AmbiqMicro/Include -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/CMSIS/ARM/Include -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/uecc/micro-ecc -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/hci/ambiq -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/hci/include -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/hci/ambiq/apollo3 -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/apps/app -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/apps/app/include -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/services -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/stack/hci -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/stack/cfg -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/sec/include -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/sec/common -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/services -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/ws-core/include -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/ws-core/sw/util -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/ws-core/sw/wsf/ambiq -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/ws-core/sw/wsf/include -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/stack/include -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/profiles -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/profiles/gatt -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/profiles/gap -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/libraries/EEPROM/src -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/libraries/PDM/src -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/libraries/RTC/src -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/libraries/Servo/src -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/libraries/SoftwareSerial/src -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/libraries/SPI/src -I/home/jrlab/.platformio/pac [... cut ...]
arm-none-eabi-ar rc .pio/build/SparkFun_RedBoard_Artemis/libvariant.a
arm-none-eabi-ranlib .pio/build/SparkFun_RedBoard_Artemis/libapollo3_sdk_mcu.a
arm-none-eabi-ranlib .pio/build/SparkFun_RedBoard_Artemis/libvariant.a
arm-none-eabi-ranlib .pio/build/SparkFun_RedBoard_Artemis/libFrameworkArduino.a
arm-none-eabi-ranlib .pio/build/SparkFun_RedBoard_Artemis/libapollo3_sdk_utils.a
arm-none-eabi-ranlib .pio/build/SparkFun_RedBoard_Artemis/libapollo3_sdk_devices.a
src/main.cpp:1:10: fatal error: Arduino.h: No such file or directory
```

jerabaul29, 2020-12-05:
So the current status is that:

- I had to set the port by hand to be able to start compiling. No big deal, but would be great to show in the readme how to only build, not try to build and upload

- Given this, I can start the compilation, but it aborts very soon because it cannot find some Arduino.h headers. Any idea how to fix this?

jerabaul29, 2020-12-05:
I think the problem with the Arduino.h is specific to this "core", as I am able to compile with the same platformio on the same computer an Arduino Mega project that also starts by importing Arduino.h.

maxgerhardt, 2020-12-05:
If it can't find the Arduino core then in the "Installation" section of the README something went wrong when cloning the Arduino core. Does `~/.platformio/packages/framework-arduinoapollo3` exist?

I've also forked the current repository and added the `package.json` to it, you may be able to write 

```
platform_packages = 
   framework-arduinoapollo3@https://github.com/maxgerhardt/Arduino_Apollo3.git
```
in the `platformio.ini` to download it.

jerabaul29, 2020-12-05:
I would think everything went fine:

```
~$ cd ~/.platformio/packages/framework-arduinoapollo3
~/.platformio/packages/framework-arduinoapollo3 [master|…1]> git status
On branch master
Your branch is up to date with 'origin/master'.

Untracked files:
  (use "git add <file>..." to include in what will be committed)

	package.json

nothing added to commit but untracked files present (use "git add" to track)
~/.platformio/packages/framework-arduinoapollo3 [master|…1]> cat package.json 
{
    "name": "framework-arduinoapollo3",
    "description": "Arduino Wiring-based Framework (Apollo3 Core)",
    "version": "1.0.22",
    "url": "https://github.com/sparkfun/Arduino_Apollo3"
}
```

maxgerhardt, 2020-12-05:
When you start compilation / `pio run`, what are the first like 20 lines of the output? It should show all packages and versions at the start. If `framework-arduinoapollo3` is not in there, that's a problem.

jerabaul29, 2020-12-05:
This is the full output I get:

```
~/Desktop/Current/test_artemis$ platformio run -v --upload-port ttyUSB0
Processing SparkFun_Artemis_Nano (platform: apollo3blue; board: SparkFun_Artemis_Nano; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/apollo3blue/SparkFun_Artemis_Nano.html
PLATFORM: Apollo 3 Blue (0.0.1) > SparkFun RedBoard Artemis Nano
HARDWARE: AMA3B1KK 48MHz, 348KB RAM, 937.50KB Flash
PACKAGES: 
 - framework-arduinoapollo3 1.0.22 
 - toolchain-gccarmnoneeabi 1.90301.200702 (9.3.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
arm-none-eabi-g++ -o .pio/build/SparkFun_Artemis_Nano/src/test_artemis.cpp.o -c -std=gnu++11 -std=gnu++11 -fno-threadsafe-statics -fno-rtti -c -g -MMD -mthumb -mfloat-abi=hard -fdata-sections -Os -ffunction-sections -nostdlib --param max-inline-insns-single=500 -fno-exceptions -mcpu=cortex-m4 -DPART_apollo3 -DAM_PACKAGE_BGA -DAM_PART_APOLLO3 -DARDUINO_ARCH_APOLLO3 -DPLATFORMIO=50004 -DARDUINO_AM_AP3_SFE_BB_ARTEMIS_NANO -DF_CPU=48000000L -DARDUINO=10809 -Iinclude -Isrc -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis_nano/config -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis_nano/bsp -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/ard_sup -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/ard_sup/ard_supers -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis_nano -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/mcu/apollo3 -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/mcu/apollo3/hal -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/mcu/apollo3/regs -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/utils -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/devices -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/CMSIS/AmbiqMicro/Include -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/CMSIS/ARM/Include -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/uecc/micro-ecc -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/hci/ambiq -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/hci/include -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/hci/ambiq/apollo3 -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/apps/app -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/apps/app/include -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/services -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/stack/hci -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/stack/cfg -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/sec/include -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/sec/common -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/services -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/ws-core/include -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/ws-core/sw/util -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/ws-core/sw/wsf/ambiq -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/ws-core/sw/wsf/include -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/stack/include -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/profiles -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/profiles/gatt -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/profiles/gap -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/libraries/EEPROM/src -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/libraries/PDM/src -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/libraries/RTC/src -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/libraries/Servo/src -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/libraries/SoftwareSerial/src -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/libraries/SPI/src -I/home/jrlab/.platfo [... cut ...] ies/Servo/src -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/libraries/SoftwareSerial/src -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/libraries/SPI/src -I/home/jrlab/.platformio/packages/framework-arduinoapollo3/libraries/Wire/src /home/jrlab/.platformio/packages/framework-arduinoapollo3/libraries/PDM/src/PDM.cpp
============================================================== [FAILED] Took 0.34 seconds ==============================================================
```

maxgerhardt, 2020-12-05:
The `Arduino.h` is in a submodule of the Arduino framework (see https://github.com/maxgerhardt/Arduino_Apollo3/tree/master/cores/arduino and https://github.com/sparkfun/arduino-mbed-bridge/tree/b54bfcb77175565cdd2fc0015f8c8f922b6cd3de). 

Are you sure you did a recursive clone? If not, please do a `cd ~/.platformio/packages/framework-arduinoapollo3/ && git submodule init && git submodule update`

jerabaul29, 2020-12-05:
Ahhh, that may be the explanation. I just copy pasted the instruction from the readme:

```
~/.platformio>/packages> git clone https://github.com/sparkfun/Arduino_Apollo3.git framework-arduinoapollo3
```

Do you want to add the submodule commands there? :)

maxgerhardt, 2020-12-05:
Ah, I see I'm looking at the most recent version which corresponds to the v2 version where they majorly reworked it with mbed-os.. The author worked, it seems, with an ealier version `1.0.22`, corresponding to I think https://github.com/sparkfun/Arduino_Apollo3/releases/tag/v1.0.22_ble-beta. Maybe that specific tag should be checked out instead. 

If that doesn't work I'll try and get a working setup myself.

jerabaul29, 2020-12-05:
Ok, so you mean all this should be done against the branch Core_V2 of this repo, right? Or you mean that I should git submodule clone a specific version of the Arduino repo? Or both? ^^ :)

jerabaul29, 2020-12-05:
Tried with the git submodule cloning, still problems.

jerabaul29, 2020-12-05:
Mmmh, if you have time, I wonder if it would be best that you have a look at how to fix this "reliably" :) It increases quite a bit the threshold for using it when following the readme on this repo does not give a working installation :) . I am not too experienced with this sort of dependencies issues, will try a bit more but if I cannot get it to work within an hour or so wait for an update of this repo :) .

maxgerhardt, 2020-12-05:
I'll definitely check this out now and let you know. But I think there's just a version decrepeancy here.

maxgerhardt, 2020-12-05:
Ah there it's all written already: https://github.com/nigelb/platform-apollo3blue/issues/5#issuecomment-728388373

jerabaul29, 2020-12-05:
Yippee, many thanks :) . Ok, that would be nice if there was an easy fix :) .

jerabaul29, 2020-12-05:
Ooh, sorry I missed that, my bad. I had a quick look at the other issues when I started on this one, but then it was another problem I had. Would be great if this could be fixed 'once and for all'. Sorry again for the noise here...

maxgerhardt, 2020-12-05:
I can get the project to compile after following the steps linked above there. Then I create a new directory and project with `mkdir artemis && cd artemis && pio init -b SparkFun_Artemis_Nano`, add the `upload_port = COM1` dummy parameter and that compiles then.

```
pio run
Processing artemis (platform: apollo3blue; board: SparkFun_Artemis_Nano; framework: arduino)
------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/apollo3blue/SparkFun_Artemis_Nano.html
PLATFORM: Apollo 3 Blue (0.0.1) > SparkFun RedBoard Artemis Nano
HARDWARE: AMA3B1KK 48MHz, 348KB RAM, 937.50KB Flash
PACKAGES:
 - framework-arduinoapollo3 1.0.23
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\artemis\src\main.cpp.o
[..]
Linking .pio\build\artemis\program.exe
Checking size .pio\build\artemis\program.exe
arm-none-eabi-objcopy -O binary .pio\build\artemis\program.exe .pio\build\artemis\firmware.bin
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  23.0% (used 81924 bytes from 356352 bytes)
Flash: [          ]   0.1% (used 988 bytes from 960000 bytes)
============================================= [SUCCESS] Took 4.17 seconds =============================================
```

jerabaul29, 2020-12-05:
Amazing, I will try it tonight :)

Any idea how to make this into a 'turn key' experience?

maxgerhardt, 2020-12-05:
Yes absolutely, the packages can be directly uploaded in their correct version to the PlatformIO package repository which the Platform can just refer to; Also the entire platform can be uploaded there. Ivan already suggested this in #8 so with the maintainer's help this can be done.

jerabaul29, 2020-12-05:
As we now understand where the problem was, and it is actually covered in other issues, closing. Many thanks again for your help :) .

closed 2020-12-05 by jerabaul29