nigelb/platform-apollo3blue #8: Simplify installation procedure

ivankravets, 2020-12-05:
Hi @nigelb,

This is Ivan from PlatformIO! Thanks for providing this integration for Apollo!

We can significantly simplify the installation guide by publishing packages into the PlatformIO Registry.

Could you create an account following https://docs.platformio.org/en/latest/plus/pio-account.html?

Also, is this dev-platform compatible with the 2.0 version of the Arduino framework?

nigelb, 2021-01-22:
Hi @ivankravets, thanks for the instructions. I will gave it look once I get everything working with the Core_V2 branch.

jerabaul29, 2021-08-31:
Just being curious: what is the status on this point? :) .

nigelb, 2021-08-31:
Hi @jerabaul29 I wrote a bit about this a while back [here](https://github.com/sparkfun/Arduino_Apollo3/issues/14#issuecomment-859583332).