nigelb/platform-apollo3blue #9: "Arduino" vs "RTOS" upstream as 2 board "flavors"?

jerabaul29, 2020-12-05:
So if I understand well, there are 2 "main versions" of the upstream sparkfun tool: one (v1.x.x) that is a "pure Arduino core", and one (v2.x.x) that is actually on top of mbed-os RTOS, right? Do you think it would be possible ultimately to get the possibility to choose between these 2 kinds of flavors? :) .

closed 2021-06-10 by nigelb

jerabaul29, 2021-06-10:
(I see this is named in the readme but not showed explicitly; do you think you could update the Readme to show how to 1) install the two cores side by side, 2) switch between one and the other? :) ).

nigelb, 2021-06-10:
Ok have a look [here](/nigelb/platform-apollo3blue/blob/master/docs/UsingMultipleVersionsOfArduino_Apollo3.md) for a detailed example.