nigelb/platform-apollo3blue #12: Issue with Core_V2 : wrong variant pathes?

jerabaul29, 2021-01-19:
Trying to use the Core_V2 for working with the MbedOS core, I get some compilation errors:

```
> Executing task: pio run <

Processing SparkFun_RedBoard_Artemis (platform: apollo3blue; board: SparkFun_RedBoard_Artemis; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/apollo3blue/SparkFun_RedBoard_Artemis.html
PLATFORM: Apollo 3 Blue (0.0.1) > SparkFun RedBoard Artemis
HARDWARE: AMA3B1KK 48MHz, 348KB RAM, 937.50KB Flash
PACKAGES: 
 - framework-arduinoapollo3 2.0.2 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
-------------------> /home/jrmet/.platformio/packages/framework-arduinoapollo3/cores/arduino/mbed-bridge
-------------------> <SCons.Script.SConscript.SConsEnvironment object at 0x7f19aa0dc490>
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/SparkFun_RedBoard_Artemis/src/main.cpp.o
arm-none-eabi-g++: error: /home/jrmet/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis/mbed/.cxx-flags: No such file or directory
arm-none-eabi-g++: error: /home/jrmet/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis/mbed/.includes: No such file or directory
arm-none-eabi-g++: error: /home/jrmet/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis/mbed/.cxx-symbols: No such file or directory
Compiling .pio/build/SparkFun_RedBoard_Artemis/mbed_bridge/bridge/pins.cpp.o
arm-none-eabi-g++: error: /home/jrmet/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis/mbed/.cxx-flags: No such file or directory
arm-none-eabi-g++: error: /home/jrmet/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis/mbed/.includes: No such file or directory
arm-none-eabi-g++: error: /home/jrmet/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis/mbed/.cxx-symbols: No such file or directory
Compiling .pio/build/SparkFun_RedBoard_Artemis/mbed_bridge/core-api/api/Common.cpp.o
arm-none-eabi-g++: error: /home/jrmet/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis/mbed/.cxx-flags: No such file or directory
arm-none-eabi-g++: error: /home/jrmet/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis/mbed/.includes: No such file or directory
arm-none-eabi-g++: error: /home/jrmet/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis/mbed/.cxx-symbols: No such file or directory
Compiling .pio/build/SparkFun_RedBoard_Artemis/mbed_bridge/core-api/api/IPAddress.cpp.o
arm-none-eabi-g++: error: /home/jrmet/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis/mbed/.cxx-flags: No such file or directory
arm-none-eabi-g++: error: /home/jrmet/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis/mbed/.includes: No such file or directory
arm-none-eabi-g++: error: /home/jrmet/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis/mbed/.cxx-symbols: No such file or directory
Compiling .pio/build/SparkFun_RedBoard_Artemis/mbed_bridge/core-api/api/PluggableUSB.cpp.o
arm-none-eabi-g++: error: /home/jrmet/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis/mbed/.cxx-flags: No such file or directory
arm-none-eabi-g++: error: /home/jrmet/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis/mbed/.includes: No such file or directory
arm-none-eabi-g++: error: /home/jrmet/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis/mbed/.cxx-symbols: No such file or directory
Compiling .pio/build/SparkFun_RedBoard_Artemis/mbed_bridge/core-api/api/Print.cpp.o
arm-none-eabi-g++: error: /home/jrmet/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis/mbed/.cxx-flags: No such file or directory
arm-none-eabi-g++: error: /home/jrmet/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis/mbed/.includes: No such file or directory
arm-none-eabi-g++: error: /home/jrmet/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis/mbed/.cxx-symbols: No such file or directory
Compiling .pio/build/SparkFun_RedBoard_Artemis/mbed_bridge/core-api/api/Stream.cpp.o
arm-none-eabi-g++: error: /home/jrmet/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis/mbed/.cxx-flags: No such file or directory
arm-none-eabi-g++: error: /home/jrmet/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis/mbed/.includes: No such file or directory
arm-none-eabi-g++: error: /home/jrmet/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis/mbed/.cxx-symbols: No such file or directory
*** [.pio/build/SparkFun_RedBoard_Artemis/src/main.cpp.o] Error 1
*** [.pio/build/SparkFun_RedBoard_Artemis/mbed_bridge/bridge/pins.cpp.o] Error 1
*** [.pio/build/SparkFun_RedBoard_Artemis/mbed_bridge/core-api/api/Common.cpp.o] Error 1
*** [.pio/build/SparkFun_RedBoard_Artemis/mbed_bridge/core-api/api/IPAddress.cpp.o] Error 1
*** [.pio/build/SparkFun_RedBoard_Artemis/mbed_bridge/core-api/api/PluggableUSB.cpp.o] Error 1
*** [.pio/build/SparkFun_RedBoard_Artemis/mbed_bridge/core-api/api/Print.cpp.o] Error 1
*** [.pio/build/SparkFun_RedBoard_Artemis/mbed_bridge/core-api/api/Stream.cpp.o] Error 1
Compiling .pio/build/SparkFun_RedBoard_Artemis/mbed_bridge/core-api/api/String.cpp.o
arm-none-eabi-g++: error: /home/jrmet/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis/mbed/.cxx-flags: No such file or directory
arm-none-eabi-g++: error: /home/jrmet/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis/mbed/.includes: No such file or directory
arm-none-eabi-g++: error: /home/jrmet/.platformio/packages/framework-arduinoapollo3/variants/redboard_artemis/mbed/.cxx-symbols: No such file or directory
*** [.pio/build/SparkFun_RedBoard_Artemis/mbed_bridge/core-api/api/String.cpp.o] Error 1
============================================================================= [FAILED] Took 0.71 seconds =============================================================================
The terminal process "pio 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

I think the problem comes from the fact that the paths actually look like:

```
~/.platformio/packages/framework-arduinoapollo3/variants [master|…1]> ls
SFE_ARTEMIS      SFE_ARTEMIS_DK     SFE_ARTEMIS_NANO        SFE_EDGE
SFE_ARTEMIS_ATP  SFE_ARTEMIS_MM_PB  SFE_ARTEMIS_THING_PLUS  SFE_EDGE2
```

I.e. ```SFE_ARTEMIS_DF``` instead of ```redboard_artemis``` as a variant in the path.

jerabaul29, 2021-01-19:
Confirm that this was the problem; by modifying this:

https://github.com/nigelb/platform-apollo3blue/blob/75f9447ce6342d18dc69b720024353811aea0134/boards/SparkFun_RedBoard_Artemis.json#L6

Into:

```    "variant": "SFE_ARTEMIS_DK",```

the compilation advances much further.

nigelb, 2021-01-22:
@jerabaul29 I fixed this with the same commit as issue #13

jerabaul29, 2021-01-22:
Yes, looks good now, many thanks, closing.

closed 2021-01-22 by jerabaul29