nigelb/platform-apollo3blue #13: Issue with Core_V2: wrong linker script file

jerabaul29, 2021-01-19:
When applying the fixes described in issues #11 #12 , I still get this error:

```
/home/jrmet/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: cannot open linker script file /home/jrmet/.platformio/packages/framework-arduinoapollo3/tools/uploaders/asb/artemis_sbl_svl_app.ld: No such file or directory
collect2: error: ld returned 1 exit status
*** [.pio/build/SparkFun_RedBoard_Artemis/program] Error 1
```

The corresponding asb folder (a git submodule of the sparkfun apollo3 code) contains indeed:

```
~/.platformio/packages/framework-arduinoapollo3/tools/uploaders/asb [:fc2bc19|✔]> ls
0xC000.ld  am_defines.py  asb.py  dist  keys_info.py  README.md  requirements.txt
```

Does that mean that the linker script should be ```0xC000.ld``` instead?

Another thing: it looks like the asb bootlader does not work any more, and that Sparkfun now recommends to use the svl bootloader, see for example:

https://github.com/sparkfun/Arduino_Apollo3/issues/310

Should this package too change to using the SVL, with baudrate 460800 ?

jerabaul29, 2021-01-19:
I think these issues are not with the linked script though, but a new issue (version of compiler or compiler used or something in this kind?).

jerabaul29, 2021-01-19:
Ok, I have been playing around, this seems well to be related to 1) finding the right linker script, 2) using the right flags on it. Cannot get it to compile yet though.

jerabaul29, 2021-01-19:
Ok, looking at the output from the Sparkfun Arduino IDE when successfully compiling a script should provide the flags and working linker file I suppose, right?

```
Linking everything together...
/home/jrlab/.arduino15/packages/SparkFun/tools/arm-none-eabi-gcc/8-2018-q4-major/bin/arm-none-eabi-gcc -T/home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/tools/uploaders/svl/0x10000.ld -Wl,-Map,/tmp/arduino_build_354051/recipe_RTC.ino.map -o /tmp/arduino_build_354051/recipe_RTC.ino.axf /tmp/arduino_build_354051/sketch/recipe_RTC.ino.cpp.o /tmp/arduino_build_354051/core/config/pins.cpp.o /tmp/arduino_build_354051/core/variant.cpp.o -Wl,--whole-archive /tmp/arduino_build_354051/core/core.a -Wl,--no-whole-archive -Wl,--whole-archive /home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/variants/SFE_ARTEMIS/mbed/libmbed-os.a -Wl,--no-whole-archive @/home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/variants/SFE_ARTEMIS/mbed/.ld-flags --specs=nano.specs -lsupc++ -lstdc++ -lm -DARDUINO=10813 -DARDUINO_APOLLO3_SFE_ARTEMIS -DARDUINO_ARCH_MBED -DARDUINO_ARCH_APOLLO3 -DMBED_NO_GLOBAL_USING_DIRECTIVE -DCORDIO_ZERO_COPY_HCI @/home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/variants/SFE_ARTEMIS/mbed/.ld-symbols
```

nigelb, 2021-01-22:
Hi @jerabaul29 I hadn't updated the boards which has the linker script name. 
I had also been working on other linking issues which I have committed.
Have a go of the latest commits.

jerabaul29, 2021-01-22:
Amazing, works fine now :) Feel free to close.

A side note: I suppose that these flags are just copied from the Sparkfun file(s) with linker flags (and other flags needed to compile build etc), right? Something like the flags for the ld script here getting copied, right?

https://github.com/sparkfun/Arduino_Apollo3/tree/master/variants/SFE_ARTEMIS_DK/mbed

Do you think it would be possible to add an "expert readme" or something like this that lists where these flags come from (ie Sparkfun), and where in your code they get copied, so that users can easily pull themselves updated flags from Sparkfun between your releases? :)

I was even wondering if there would be some possibility to programmatically read these flags from the Sparkfun files, so that these get always up to date by just pulling the sparkfun repo? May be material for another issue though, let me know if I should open a new one just to keep track.

closed 2021-06-07 by nigelb