nigelb/platform-apollo3blue #14: issue with the linter not being able to find the type of rtos variables

jerabaul29, 2021-01-19:
It seems that using installation following the fixed steps as described in #11 #12 the linter / intellisense has difficulties finding the type of rtos / mbed variables, see screenshot under. Any idea how to fix? (see <error-type> thread_1 hoover pop up.

[image]

nigelb, 2021-01-22:
Hi @jerabaul29 I suspect it has to do with the way the `c`,`cxx`,`ld` flags are done in sparkfun/Arduino_Apollo3 now: 

https://github.com/sparkfun/Arduino_Apollo3/blob/58fb804e7865e76e01ef52a7691cc5f658b032d7/platform.txt#L47

```
compiler.c.flags=-iprefix{runtime.platform.path}/cores/ @{build.variant.path}/mbed/.c-flags {compiler.c.extra_flags}
```

which I did the same way:

https://github.com/nigelb/platform-apollo3blue/blob/a3e7b2d8444fd8d9920c455fe54a4b1d50dbff70/builder/frameworks/arduino.py#L53-L60 

My plan is to revisit this once everything else is working properly and read in the file and add the flags into the various lists

jerabaul29, 2021-01-22:
Yes, seems to work perfectly now, amazing, many thanks. Feel free to close if you want (I do not want to close myself as you named you wanted to revisit / test) .