nigelb/platform-apollo3blue #15: sketches use surprisingly large amounts of RAM

jerabaul29, 2021-01-22:
I am playing around with some small Artemis + MbedOS recipes. I have a small recipe I wanted to try (available at https://github.com/jerabaul29/Artemis_MbedOS_recipes/tree/main/recipes/recipe_RTC if you want to reproduce).

I am able to compile it with the Arduino IDE (well, not with the latest Sparkfun release, see issue https://github.com/sparkfun/Arduino_Apollo3/issues/330 , but with the previous release from a few weeks ago it works fine), in which case the sketch uses a few percents of the RAM available.

But when I compile this exact sketch with the platformio extension, I get:

```
Processing SparkFun_RedBoard_Artemis (platform: apollo3blue; board: SparkFun_RedBoard_Artemis; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/apollo3blue/SparkFun_RedBoard_Artemis.html
PLATFORM: Apollo 3 Blue (0.0.2) > SparkFun RedBoard Artemis
HARDWARE: AMA3B1KK 48MHz, 384KB RAM, 960KB Flash
PACKAGES: 
 - framework-arduinoapollo3 2.0.2 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
-------------------> /home/jrmet/.platformio/packages/framework-arduinoapollo3/cores/arduino/mbed-bridge
-------------------> <SCons.Script.SConscript.SConsEnvironment object at 0x7fc005b98520>
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/SparkFun_RedBoard_Artemis/src/main.cpp.o
Compiling .pio/build/SparkFun_RedBoard_Artemis/varient/config/pins.cpp.o
Compiling .pio/build/SparkFun_RedBoard_Artemis/varient/variant.cpp.o
Compiling .pio/build/SparkFun_RedBoard_Artemis/mbed_bridge/bridge/pins.cpp.o
Compiling .pio/build/SparkFun_RedBoard_Artemis/mbed_bridge/core-api/api/Common.cpp.o
Compiling .pio/build/SparkFun_RedBoard_Artemis/mbed_bridge/core-api/api/IPAddress.cpp.o
Compiling .pio/build/SparkFun_RedBoard_Artemis/mbed_bridge/core-api/api/PluggableUSB.cpp.o
Compiling .pio/build/SparkFun_RedBoard_Artemis/mbed_bridge/core-api/api/Print.cpp.o
Compiling .pio/build/SparkFun_RedBoard_Artemis/mbed_bridge/core-api/api/Stream.cpp.o
src/main.cpp: In function 'int main()':
src/main.cpp:13:37: warning: 'void rtos::ThisThread::sleep_for(uint32_t)' is deprecated: Pass a chrono duration, not an integer millisecond count. For example use `5s` rather than `5000`. [since mbed-os-6.0.0] [-Wdeprecated-declarations]
     rtos::ThisThread::sleep_for(1000);
                                     ^
In file included from /home/jrmet/.platformio/packages/framework-arduinoapollo3/cores/mbed-os/rtos/rtos.h:30:0,
                 from /home/jrmet/.platformio/packages/framework-arduinoapollo3/cores/mbed-os/mbed.h:22,
                 from /home/jrmet/.platformio/packages/framework-arduinoapollo3/cores/arduino/mbed-bridge/Arduino.h:11,
                 from /home/jrmet/.platformio/packages/framework-arduinoapollo3/cores/arduino/sdk/ArduinoSDK.h:9,
                 from <command-line>:0:
/home/jrmet/.platformio/packages/framework-arduinoapollo3/cores/mbed-os/rtos/ThisThread.h:216:6: note: declared here
 void sleep_for(uint32_t millisec);
      ^~~~~~~~~
src/main.cpp:115:41: warning: 'void rtos::ThisThread::sleep_for(uint32_t)' is deprecated: Pass a chrono duration, not an integer millisecond count. For example use `5s` rather than `5000`. [since mbed-os-6.0.0] [-Wdeprecated-declarations]
         rtos::ThisThread::sleep_for(1000);
                                         ^
In file included from /home/jrmet/.platformio/packages/framework-arduinoapollo3/cores/mbed-os/rtos/rtos.h:30:0,
                 from /home/jrmet/.platformio/packages/framework-arduinoapollo3/cores/mbed-os/mbed.h:22,
                 from /home/jrmet/.platformio/packages/framework-arduinoapollo3/cores/arduino/mbed-bridge/Arduino.h:11,
                 from /home/jrmet/.platformio/packages/framework-arduinoapollo3/cores/arduino/sdk/ArduinoSDK.h:9,
                 from <command-line>:0:
/home/jrmet/.platformio/packages/framework-arduinoapollo3/cores/mbed-os/rtos/ThisThread.h:216:6: note: declared here
 void sleep_for(uint32_t millisec);
      ^~~~~~~~~
Compiling .pio/build/SparkFun_RedBoard_Artemis/mbed_bridge/core-api/api/String.cpp.o
Archiving .pio/build/SparkFun_RedBoard_Artemis/libvarient.a
Indexing .pio/build/SparkFun_RedBoard_Artemis/libvarient.a
Compiling .pio/build/SparkFun_RedBoard_Artemis/mbed_bridge/core-implement/CommonAnalog.cpp.o
Compiling .pio/build/SparkFun_RedBoard_Artemis/mbed_bridge/core-implement/CommonDigital.cpp.o
Compiling .pio/build/SparkFun_RedBoard_Artemis/mbed_bridge/core-implement/CommonInit.cpp.o
Compiling .pio/build/SparkFun_RedBoard_Artemis/mbed_bridge/core-implement/CommonInterrupt.cpp.o
Compiling .pio/build/SparkFun_RedBoard_Artemis/mbed_bridge/core-implement/CommonMath.cpp.o
Compiling .pio/build/SparkFun_RedBoard_Artemis/mbed_bridge/core-implement/CommonPulse.cpp.o
Compiling .pio/build/SparkFun_RedBoard_Artemis/mbed_bridge/core-implement/CommonTiming.cpp.o
Compiling .pio/build/SparkFun_RedBoard_Artemis/mbed_bridge/core-implement/HardwareSerial.cpp.o
Compiling .pio/build/SparkFun_RedBoard_Artemis/mbed_bridge/core-implement/Yield.cpp.o
Compiling .pio/build/SparkFun_RedBoard_Artemis/mbed_bridge/main.cpp.o
Compiling .pio/build/SparkFun_RedBoard_Artemis/PDM/PDM.cpp.o
[... 31 lines cut ...]
               ^~
Archiving .pio/build/SparkFun_RedBoard_Artemis/libServo.a
Indexing .pio/build/SparkFun_RedBoard_Artemis/libServo.a
Archiving .pio/build/SparkFun_RedBoard_Artemis/libPDM.a
Indexing .pio/build/SparkFun_RedBoard_Artemis/libPDM.a
Archiving .pio/build/SparkFun_RedBoard_Artemis/libmbed_bridge.a
Indexing .pio/build/SparkFun_RedBoard_Artemis/libmbed_bridge.a
Archiving .pio/build/SparkFun_RedBoard_Artemis/libRTC.a
Indexing .pio/build/SparkFun_RedBoard_Artemis/libRTC.a
Archiving .pio/build/SparkFun_RedBoard_Artemis/libSPI.a
Indexing .pio/build/SparkFun_RedBoard_Artemis/libSPI.a
Archiving .pio/build/SparkFun_RedBoard_Artemis/libWire.a
Indexing .pio/build/SparkFun_RedBoard_Artemis/libWire.a
Linking .pio/build/SparkFun_RedBoard_Artemis/program
Checking size .pio/build/SparkFun_RedBoard_Artemis/program
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==========]  99.7% (used 391928 bytes from 393216 bytes)
arm-none-eabi-objcopy -O binary .pio/build/SparkFun_RedBoard_Artemis/program .pio/build/SparkFun_RedBoard_Artemis/firmware.bin
Flash: [=         ]  12.8% (used 125592 bytes from 983040 bytes)
=========================================================================================== [SUCCESS] Took 10.83 seconds ===========================================================================================
```

And here the RAM use shocks me: 99.7 percents.

Any idea if there is a problem / why / how to solve?

jerabaul29, 2021-01-22:
I got some help compiling from the Sparkfun people :) .

I can confirm that when compiling on the Arduino IDE, the RAM use looks much different. Here the output from the Arduino IDE:

```
Sketch uses 128184 bytes (13%) of program storage space. Maximum is 983040 bytes.
Global variables use 27928 bytes (7%) of dynamic memory, leaving 365288 bytes for local variables. Maximum is 393216 bytes.
```

I.e. the Flash use seems similar between Arduino IDE and Platformio, but RAM use is 7% with Arduino, 99.7% with the Platformio build.

Any idea why / how to fix? :) .

nigelb, 2021-06-07:
Hi @jerabaul29 I have fixed the reported memory size issue and merged the Core_V2 branch to master.

closed 2021-06-10 by nigelb