nigelb/platform-apollo3blue #16: problem with the upload_port when creating new project

jerabaul29, 2021-01-23:
I always get (some variations of) the following error message when creating a new project:

```
PIO Core Call Error: "The current working directory /home/jrlab/Documents/PlatformIO/Projects/test_err_msg will be used for the 
project.\n\nThe next files/directories have been created in /home/jrlab/Documents/PlatformIO/Projects/test_err_msg\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nError: Processing SparkFun_RedBoard_Artemis (platform: apollo3blue; board: SparkFun_RedBoard_Artemis; framework: arduino)\n--------------------------------------------------------------------------------\nTool Manager: Could not parse manifest -> Expecting property name enclosed in double quotes: line 7 column 1 (char 204)\nTool Manager: Could not parse manifest -> Expecting property name enclosed in double quotes: line 7 column 1 (char 204)\nVerbose mode can be enabled via `-v, --verbose` option\nTool Manager: Could not parse manifest -> Expecting property name enclosed in double quotes: line 7 column 1 (char 204)\nError: Please specify `upload_port` for environment or use global `--upload-port` option.\nFor some development platforms it can be a USB flash drive (i.e. /media/<user>/<device name>)\n========================== [FAILED] Took 0.23 seconds =========================="
```

Basically, complaining that I need to specify the ```upload_port``` in the platformio.ini config file. Easy enough to fix, but annoying, and may scare people who are "even n00ber than me" :) .

- any idea how to fix this by default? Maybe filling with a default value at creation?
- more fundamentally, any idea why this is triggered? Should be possible to have a project without upload port, for example if the project is just for testing some compilation etc without uploading.

nigelb, 2021-06-08:
Hi @jerabaul29 this should be fixed now.

closed 2021-06-10 by nigelb

wzqvip, 2023-05-22:
config:
[env:Ambiq_Apollo3_Blue_EVB]
platform = apollo3blue
board = Ambiq_Apollo3_Blue_EVB
framework = ambiqsdk-sfe
platform_packages = framework-ambiqsuitesdkapollo3-sfe@2.5.1

I want to use this PIO to code with  Apollo 3 EVB (official)

`Processing Ambiq_Apollo3_Blue_EVB (platform: apollo3blue; board: Ambiq_Apollo3_Blue_EVB; framework: ambiqsdk-sfe)
----------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/apollo3blue/Ambiq_Apollo3_Blue_EVB.html
PLATFORM: Apollo 3 Blue (0.0.2+sha.cdbb76e) > Ambiq Apollo3 Blue EVB
HARDWARE: AMA3B1KK 48MHz, 384KB RAM, 960KB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES: 
 - framework-ambiqsuitesdkapollo3-sfe @ 2.5.1 
 - toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)
ManifestParserError: Could not parse manifest -> Invalid \escape: line 11 column 37 (char 255):
  File "C:\Users\Administrator\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 174:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Administrator\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Administrator\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Administrator\.platformio\platforms\apollo3blue\builder\main.py", line 108:
    target_elf = env.BuildProgram()
  File "C:\Users\Administrator\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Administrator\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 60:
    env.ProcessProgramDeps()
  File "C:\Users\Administrator\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Administrator\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 120:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "C:\Users\Administrator\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Administrator\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 331:
    SConscript(env.GetFrameworkScript(name), exports="env")
  File "C:\Users\Administrator\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 662:
    return method(*args, **kw)
  File "C:\Users\Administrator\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Administrator\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Administrator\.platformio\platforms\apollo3blue\builder\frameworks\ambiqsuitesdk-sfe.py", line 275:
    lib_builders.append(create_lib(ambiq_lib['src_path'], ambiq_lib['manifest_path'], context))
  File "C:\Users\Administrator\.platformio\platforms\apollo3blue\builder\frameworks\ambiqsuitesdk-sfe.py", line 107:
    manifest = ManifestParserFactory.new(manifest_data%context, ManifestFileType.from_uri(manifest_path), False)
  File "C:\Users\Administrator\.platformio\penv\lib\site-packages\platformio\package\manifest\parser.py", line 138:
    return cls(contents, remote_url, package_dir)
  File "C:\Users\Administrator\.platformio\penv\lib\site-packages\platformio\package\manifest\parser.py", line 149:
    raise ManifestParserError("Could not parse manifest -> %s" % exc) from exc
============================================================ [FAILED] Took 0.68 seconds ============================================================
`

nigelb, 2023-05-22:
Hi @wzqvip,

What command did you run to get this output?

From what I can see it looks like you may have an issue with the `toolchain-gccarmnoneeabi` package's manifest.

Have you tried deleting the package and re-running the command?
It will be in your `.platformio/packages` directory in your `$HOME` directory.
It will will be called either `toolchain-gccarmnoneeabi` or `toolchain-gccarmnoneeabi@<VERSION_NUMBER>`

Nigel.