nigelb/platform-apollo3blue #17: cannot upload to board

jerabaul29, 2021-01-23:
I have a board, with the SVL bootloader (I am able to upload from the Arduino IDE). However in Platformio I get:

```
RAM:   [==========]  99.7% (used 391928 bytes from 393216 bytes)
Flash: [=         ]  10.5% (used 103592 bytes from 983040 bytes)
Configuring upload protocol...
AVAILABLE: svl
CURRENT: upload_protocol = svl
/home/jrlab/.platformio/packages/framework-arduinoapollo3/tools/uploaders/svl/dist/linux/svl /dev/ttyUSB0 -b 921600 -f .pio/build/SparkFun_RedBoard_Artemis/firmware.bin -v

Artemis SVL Bootloader
Script version 1.7

Phase:  Setup
        Cleared startup blip
Failed to enter bootload phase

Phase:  Setup
        Cleared startup blip
Failed to enter bootload phase

Phase:  Setup
        Cleared startup blip
Failed to enter bootload phase
Target failed to enter bootload mode. Verify the right COM port is selected and that your board has the SVL bootloader.
=================================================================================== [SUCCESS] Took 2.79 seconds ===================================================================================
```

- any idea where the problem is? I have double checked the ```platformio.ini```, I think it looks good:

```
[env:SparkFun_RedBoard_Artemis]
platform = apollo3blue
board = SparkFun_RedBoard_Artemis
framework = arduino
upload_port = /dev/ttyUSB0
```

jerabaul29, 2021-01-23:
(the RAM usage issue of #15 is also visible, but I think this is a separate issue?).

nigelb, 2021-06-07:
Hi @jerabaul29 Check out the latest commits on master. It uploads for me now.

nigelb, 2021-06-09:
@jerabaul29, since I see you are using Linux, this may be relevant:  sparkfun/Apollo3_Uploader_SVL#4

closed 2021-06-10 by nigelb

cadeo111, 2022-03-09:
I was having this issue on macOS 12.1 with the Sparkfun ExpLoRaBLE
to resolve it I had to change the default upload baudrate from 921600 to 115200
```
[env:SparkFun_Thing_Plus_expLoRaBLE]
platform = apollo3blue
board = SparkFun_Thing_Plus_expLoRaBLE
framework = arduino
upload_speed = 115200; <- add this line
```

nigelb, 2022-03-09:
Hi @cadeo111,

Thanks for that.

To help the next person who has upload issues I have made a list of all the issues and solutions I have come across with [uploading](https://github.com/nigelb/platform-apollo3blue/blob/master/docs/platform-apollo3blue/Upload%20Issues.md) to the Artemis boards.