nigelb/platform-apollo3blue #20: board support request: SparkFun OpenLog Artemis

JrVerbiest, 2021-08-13:
So far I notice the SparkFun OpenLog Artemis [https://www.sparkfun.com/products/16832](url) is not supported. 
Is there a possibility to support this board?

nigelb, 2021-08-14:
Hi @JrVerbiest , the [SparkFun OpenLog Artemis](https://www.sparkfun.com/products/16832) does not show up as a board in Sparkfun's Arduino plugin:

[image]

What this means is that there is not a defined [variant](https://github.com/sparkfun/Arduino_Apollo3/tree/main/variants) for the OpenLog. However, since Sparkfun kindly provide the [schematic](https://cdn.sparkfun.com/assets/1/4/e/1/7/OpenLog_Artemis_9DoF_IMU_Schematic_v1_0_latest.pdf) you should be able to use the SparkFun_Artemis_Module "board" and use the schematic to figure out any board specific pins.

Hope this helps.

JrVerbiest, 2021-08-19:
Thanks.
The RedBoard Artemis ATP should work.

closed 2021-08-19 by JrVerbiest