nigelb/platform-apollo3blue #27: suprious warnings with core v1 for c_cpp_properties.json

jerabaul29, 2021-11-05:
I installed the platformio artemis core v1 (corresponding to this branch: https://github.com/sparkfun/Arduino_Apollo3/tree/v1 ), and set up my package.json following this:

```
jr@T490:~/.platformio/packages/framework-arduinoapollo3@1.2.3$ cat package.json 
{
    "name": "framework-arduinoapollo3",
    "description": "A bare metal Arduino core for Ambiq Apollo3 based boards",
    "version": "1.2.3",
    "url": "https://github.com/sparkfun/Arduino_Apollo3"
}
```

Things work mostly well, but I get quite a few spurious warnings for c_cpp_properties.json . I think these are warnings for tools that are in the core v2 but not the core v1 (putting the copy of the errors at the bottom).

- any ways that you could update the generation of the c_cpp_properties.json so that it does not include these files if the major version of the framework is 1 not 2?
- in the meantime, how would you recommend that I fix it? The c_cpp_properties.json explicitly says to not edit it by hand, but should I do it still? :)

```
[{
	"resource": "/home/jr/Desktop/Current/test_platformio_artemis/.vscode/c_cpp_properties.json",
	"owner": "/home/jr/Desktop/Current/test_platformio_artemis",
	"severity": 4,
	"message": "Cannot find \"/home/jr/.platformio/packages/framework-arduinoapollo3@1.2.3/cores/arduino/am_sdk_ap3/third_party/uecc/micro-ecc\".",
	"startLineNumber": 26,
	"startColumn": 17,
	"endLineNumber": 26,
	"endColumn": 131
},{
	"resource": "/home/jr/Desktop/Current/test_platformio_artemis/.vscode/c_cpp_properties.json",
	"owner": "/home/jr/Desktop/Current/test_platformio_artemis",
	"severity": 4,
	"message": "Cannot find \"/home/jr/.platformio/packages/framework-arduinoapollo3@1.2.3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/hci/ambiq\".",
	"startLineNumber": 27,
	"startColumn": 17,
	"endLineNumber": 27,
	"endColumn": 137
},{
	"resource": "/home/jr/Desktop/Current/test_platformio_artemis/.vscode/c_cpp_properties.json",
	"owner": "/home/jr/Desktop/Current/test_platformio_artemis",
	"severity": 4,
	"message": "Cannot find \"/home/jr/.platformio/packages/framework-arduinoapollo3@1.2.3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/hci/include\".",
	"startLineNumber": 28,
	"startColumn": 17,
	"endLineNumber": 28,
	"endColumn": 139
},{
	"resource": "/home/jr/Desktop/Current/test_platformio_artemis/.vscode/c_cpp_properties.json",
	"owner": "/home/jr/Desktop/Current/test_platformio_artemis",
	"severity": 4,
	"message": "Cannot find \"/home/jr/.platformio/packages/framework-arduinoapollo3@1.2.3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/hci/ambiq/apollo3\".",
	"startLineNumber": 29,
	"startColumn": 17,
	"endLineNumber": 29,
	"endColumn": 145
},{
	"resource": "/home/jr/Desktop/Current/test_platformio_artemis/.vscode/c_cpp_properties.json",
	"owner": "/home/jr/Desktop/Current/test_platformio_artemis",
	"severity": 4,
	"message": "Cannot find \"/home/jr/.platformio/packages/framework-arduinoapollo3@1.2.3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/apps/app\".",
	"startLineNumber": 30,
	"startColumn": 17,
	"endLineNumber": 30,
	"endColumn": 136
},{
	"resource": "/home/jr/Desktop/Current/test_platformio_artemis/.vscode/c_cpp_properties.json",
	"owner": "/home/jr/Desktop/Current/test_platformio_artemis",
	"severity": 4,
	"message": "Cannot find \"/home/jr/.platformio/packages/framework-arduinoapollo3@1.2.3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/apps/app/include\".",
	"startLineNumber": 31,
	"startColumn": 17,
[... 267 lines cut ...]
	"message": "Cannot find \"/home/jr/.platformio/packages/framework-arduinoapollo3@1.2.3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/profiles\".",
	"startLineNumber": 88,
	"startColumn": 21,
	"endLineNumber": 88,
	"endColumn": 140
},{
	"resource": "/home/jr/Desktop/Current/test_platformio_artemis/.vscode/c_cpp_properties.json",
	"owner": "/home/jr/Desktop/Current/test_platformio_artemis",
	"severity": 4,
	"message": "Cannot find \"/home/jr/.platformio/packages/framework-arduinoapollo3@1.2.3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/profiles/gatt\".",
	"startLineNumber": 89,
	"startColumn": 21,
	"endLineNumber": 89,
	"endColumn": 145
},{
	"resource": "/home/jr/Desktop/Current/test_platformio_artemis/.vscode/c_cpp_properties.json",
	"owner": "/home/jr/Desktop/Current/test_platformio_artemis",
	"severity": 4,
	"message": "Cannot find \"/home/jr/.platformio/packages/framework-arduinoapollo3@1.2.3/cores/arduino/am_sdk_ap3/third_party/exactle/sw/profiles/gap\".",
	"startLineNumber": 90,
	"startColumn": 21,
	"endLineNumber": 90,
	"endColumn": 144
}]
```

nigelb, 2021-11-17:
Hi @jerabaul29,

Try using the updated method of using core_v1 described [here](/nigelb/platform-apollo3blue/blob/master/docs/platform-apollo3blue/UsingMultipleVersionsOfArduino_Apollo3.md).

There is no longer a need to check out the v1 branch, it is all integrated into the main branch.

jerabaul29, 2021-11-17:
Aaah, thanks :)  Sorry I was confused and missed it. Maybe making this even more clear in the readme, ie that there is no longer need to tweak things by hand and that there is an out of the box solution, and showing an example already in the readme with both core v1 and core v2, would be helpful for future users? :) . I will try it tonight again :) .

nigelb, 2021-11-17:
Sorry, I misread the url and thought you checked out the V1 branch of this repo.

I will have another look.

nigelb, 2021-11-17:
have you tried updating the ide files with `pio init --ide vscode`

jerabaul29, 2021-11-17:
Ok, no worries, do you mean I was doing the correct thing actually? :) .

jerabaul29, 2021-11-17:
No, I have not ```pio init --ide vscode```, I can try this, thanks :) . Will look at it hopefully tonight otherwise will try to do it this week and let you know.

nigelb, 2021-11-17:
Ha, I now finally realise the issue you are seeing.
I don't normally use vscode and did not understand what issues you were talking about with a `c_cpp_properties.json` file.
Now I know what we are dealing with I should be able to fix the issues.

jerabaul29, 2021-11-17:
😊😇👌👍 Many thanks!

Just curious: what alternative to VSC do you use to run this and platformio?

nigelb, 2021-11-17:
For small test programs I generally use vim and for projects I have been using CLion.
However, in the last few weeks I have started playing with vscode, and quite like it.

jerabaul29, 2021-11-17:
Ok, thanks for the info. I did not know that it was possible to run platformio with vim.

VSCode is a bit slow at times, uses a lot of RAM, and not all modal edition commands are available even with the vim-mod package, but it works well enough and is becoming kind of a standard I guess :) .

nigelb, 2021-11-17:
Hi @jerabaul29, please update platform-apollo3blue and let me know if that resolves your issue.

jerabaul29, 2021-11-21:
I did a:

```
jr@T490:~/.platformio/platforms/apollo3blue$ git pull
```

and 1) the warnings are gone, 2) it still compiles fine :) . Amazing, many thanks, feel free to close :) .

closed 2021-11-24 by nigelb