nigelb/platform-apollo3blue #28: creating a new project using the "usual" platformio way fails

jerabaul29, 2021-11-05:
I have installed the core v1 and all works fine following the "create a project" instructions of the readme on this repo.

However, if I try to create a new project in the "usual platformio way", things fail with a warning:

```
PIO Core Call Error: "The current working directory /home/jr/Documents/PlatformIO/Projects/hello_artemis will be used for the project.\n\nThe next files/directories have been created in /home/jr/Documents/PlatformIO/Projects/hello_artemis\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nError: Could not find the package with 'framework-ambiqsuitesdkapollo3-sfe @ >2.4.0' requirements for your system 'linux_x86_64'"
```

See screenshot:

[image]

Any idea why / how to fix? It is already very useful to be able to create a project "by hand", but it would be even better if possible to use the default platformio method :) .

nigelb, 2021-11-17:
Hi @jerabaul29,

If you install the [AmbiqSDK](/nigelb/platform-apollo3blue/blob/master/docs/platform-apollo3blue/UsingSparkfunsAmbiqSDK.md) package, the error should go away.

It should not be required, however since I always have it installed I missed this issue.
I will look into why the package seems to be required.

jerabaul29, 2021-11-17:
Thanks for the explanation :) . I will wait for a fix before updating then, as I would like to not pull this repo - seems quite out of date and, as you say, it should not be needed in my case :) . But of course no worries and no hurry, I can live with things as they are now in the meantime :) .

jerabaul29, 2021-11-21:
I tried to install the AmbiqSDK, think I got it right:

```
jr@T490:~/.platformio/packages$ git clone --recurse-submodules --branch 2.5.1-sfe https://github.com/sparkfun/AmbiqSuiteSDK.git framework-ambiqsuitesdkapollo3-sfe@2.5.1 
jr@T490:~/.platformio/packages/framework-ambiqsuitesdkapollo3-sfe@2.5.1$ cat package.json 
{
    "name": "framework-ambiqsuitesdkapollo3-sfe",
    "description": "SparkFun's AmbiqSuiteSDK repository.",
    "version": "2.5.1",
    "url": "https://github.com/sparkfun/AmbiqSuiteSDK"
}
```

But I still have the error message popping up. Any idea? :)

PS: a small comment: on the install instructions you linked to, this: ```~/.platformio> git clone -- [...]``` should be ```~/.platformio/packages> git clone -- [...]```, right? :)

nigelb, 2021-11-23:
Hi @jerabaul29,

I can't replicate your issue. I tried on both Linux and Windows.

So that we are on the same page, this is what I did (On Windows in a git bash window):

First I deleted my .platformio directory:

    rm -Rf ~/.platformio

Then I created the relevant directories:

    $ mkdir .platformio/
    $ cd  .platformio/
    $ mkdir platforms packages

I Checked out sparkfun/Arduino_Apollo3 (I am using the new release - 2.1.1):

    $ cd ~/.platformio/packages
    $ git clone --recurse-submodules --branch v2.1.1 https://github.com/sparkfun/Arduino_Apollo3.git framework-arduinoapollo3@2.1.1

I created the `package.json` file in `~/.platformio/packages/framework-arduinoapollo3\@2.1.1/` with the contents:
````json
{
    "name": "framework-arduinoapollo3",
    "description": "An mbed-os enabled Arduino core for Ambiq Apollo3 based boards",
    "version": "2.1.1",
    "url": "https://github.com/sparkfun/Arduino_Apollo3"
}
````

I checked out platform-apollo3blue:

    $ cd ~/.platformio/platforms 
    $ git clone https://github.com/nigelb/platform-apollo3blue.git apollo3blue

Then in VSCode I press the "New Project" button:

[image]

I fill in the project wizard:

[image]

I wait for the project wizard:

[image]

I trust the project:

[image]

I have my project:

[image]

[image]

And it compiles:

[image]

jerabaul29, 2021-11-23:
Ok, interesting. I can try this tonight. Can it be that the difference is that you are using the core v2, while I am using the core v1? (I need barebone, the RTOS is too heavy for me and has too much overhead in my case).

I can try to re install tonight the core v2, check if I have the problem on the core v2, and check if I have the problem on the core v1.

nigelb, 2021-11-24:
Hi @jerabaul29 

You are correct, it does have to do with you using the core v1.
I can replicate it now, both in VSCode and on the command line.

First I deleted my .platformio directory:

    rm -Rf ~/.platformio

Then I created the relevant directories:

    $ mkdir ~/.platformio/
    $ cd  ~/.platformio/
    $ mkdir platforms packages

I Checked out sparkfun/Arduino_Apollo3 (I am using the new release - 2.1.1):

    $ cd ~/.platformio/packages
    $ git clone --recurse-submodules --branch v1.2.3 https://github.com/sparkfun/Arduino_Apollo3.git framework-arduinoapollo3@1.2.3

Modify the `package.json` file in `~/.platformio/packages/framework-arduinoapollo3\@1.2.3/` to:
````json
{
    "name": "framework-arduinoapollo3",
    "description": "Arduino Wiring-based Framework (Apollo3 Core)",
    "version": "1.2.3",
    "url": "https://github.com/sparkfun/Arduino_Apollo3"
}
````

I checked out platform-apollo3blue:

    $ cd ~/.platformio/platforms 
    $ git clone https://github.com/nigelb/platform-apollo3blue.git apollo3blue

The I create a new project:

    $ mkdir core_v1_test
    $ cd core_v1_test
    $ pio init --board SparkFun_RedBoard_Artemis_ATP

    The current working directory C:\Users\user\Documents\PlatformIO\Projects\Test_ATP_Core_V1 will be used for the project.

    The next files/directories have been created in C:\Users\user\Documents\PlatformIO\Projects\Test_ATP_Core_V1
    include - Put project header files here
    lib - Put here project specific (private) libraries
    src - Put project source files here
    platformio.ini - Project Configuration File

Then try and compile it:

    $ pio run
    Processing SparkFun_RedBoard_Artemis_ATP (platform: apollo3blue; board: SparkFun_RedBoard_Artemis_ATP; framework: arduino)
    --------------------------------------------------------------------------------

    Tool Manager: Installing framework-arduinoapollo3 @ >1
    Error: Could not find the package with 'framework-arduinoapollo3 @ >1' requirements for your system 'windows_amd64'

The issue is this: https://github.com/nigelb/platform-apollo3blue/blob/d388f9e7e49ec8fb7f077ec22defa0df3dd24618/platform.json#L36

It would seem that platformio does not consider version `1.2.3` to be greater than version `1`.

nigelb, 2021-11-24:
@jerabaul29 please update platform-apollo3blue and let me know how you go.

jerabaul29, 2021-11-24:
Many thanks for all your hard work on this :) . A bit hectic here these days, but hope to look at it within a day or two :) Thanks again :) .

jerabaul29, 2021-11-24:
Got the time to test just now, all working now, amazing, many thanks :) .

Feel free to close as far as I am concerned :) .

closed 2021-11-24 by nigelb