nigelb/platform-apollo3blue #29: brainstorming: releasing as a "all-parts-included", "turn-key" container?

jerabaul29, 2021-11-17:
Seeing that some people around me struggle a bit getting all installed correctly, and wondering: would it be useful to distribute an "all-parts-included", "turn-key" development environment as a container containing i) visual studio code ii) platformio iii) the core / board provided by this repo with all its dependencies? There could either be a single container if all works seamlessly with both core v1 and core v2, or 2 containers if there are some friction points.

nigelb, 2021-12-09:
Hi @jerabaul29, I am thinking about this. 

In the mean time I have:
1. Forked the [sparkfun/Arduino_Apollo3](https://github.com/sparkfun/Arduino_Apollo3) repo [here](https://github.com/nigelb/Arduino_Apollo3) and added the required files
2. Updated the instillation instructions (to use said repo) so for simple cases platform-apollo3blue should be easier to install now.

Installing can now be as simple as:
````
$ pio platform install git+https://github.com/nigelb/platform-apollo3blue
$ mkdir new_project
$ cd new_project
$ pio init -b SparkFun_RedBoard_Artemis_ATP -O"platform_packages=framework-arduinoapollo3@https://github.com/nigelb/Arduino_Apollo3#v2.1.1_pio"
````

jerabaul29, 2021-12-09:
Super nice :) . Hope this can be pushed to the sparkfun main repo and registered by sparkfun on the platformio registry :) .

idea--list, 2022-01-14:
@nigelb 
This repo is really awesome! As MbedOS seems to becoming abandonware i was looking for something else than Arduino IDE to program my Artemis Thing Plus. Tried the native Ambiq SDK but that has no documentation at all and i just can not afford several weeks studying the code base and writing my own notes before i can code anything useful beyond blinking some LEDs.

I knew there also exists an Arduino core, but i really dislike Arduino IDE itself as it is way too limiting and also renaming a .c or .cpp file to an .ino file is not my style of doing things. Had some experience with VS Code coding in PHP and then somehow i stumbled upon this repo. During the installation there was a point where i needed to figure out how to get it right and also it took a while to discover that the Artemis Thing Plus is referred as just Thing Plus (needed in the project initialization code).

I think @jerabaul29 is right, with a container there would be less friction for those who want to use it for the first time. The internal board names should be exactly the same as Sparkfun is selling them, not just something similar. Am happy i found your repo! 🙂

nigelb, 2022-01-14:
Hi @idea--list glad you are finding this project useful.

Do you (or @jerabaul29) know of an existing project that has a container that I can look at as an example?

I have noticed a couple of other board name inconsistencies, I was just unsure weather I should fix them and break peoples existing projects. Even though it would only take a moment for them to fix them up.

idea--list, 2022-01-14:
@nigelb 
I never made any containers before, just used some of them to get whole environments the same as other people in a small dev team... so can not help you how to build a container.
As for the board name inconsistencies: i can truly understand your dilemma, however the number of people who might begin using this repo is greater than that of those who are already using it. Either way a clear documentation of the different board names would be beneficial. Maybe the best option would be to fix the names and also put some text inside the warning messages if one uses the old alias when initializing a new project or when building fails of an existing project.

tbjers, 2022-01-14:
Do we need the container if this can just be adapted as another platform/framework/whatever directly in PlatformIO? I have run VS Code in Snaps before, and it always end up annoying me because of the process isolation. I'd prefer to keep the editor and everything else out of the process.