nigelb/platform-apollo3blue #30: attempting to compile compiles but also tries to upload (and fails if no board available)

jerabaul29, 2021-11-21:
Clicking on the "Platformio: Build" button in the VSC bottom bar compiles the code fine as expected, but it also tries to upload (not as expected). As a consequence, if no board is available, there may be situations like i) compile works fine, but ii) upload fails and the command returns an error status code. See:

```
Building in release mode
`.pio/build/SparkFun_RedBoard_Artemis/firmware.bin' is up to date.
Checking size .pio/build/SparkFun_RedBoard_Artemis/program
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  29.4% (used 115716 bytes from 393216 bytes)
Flash: [          ]   0.8% (used 7384 bytes from 983040 bytes)
Configuring upload protocol...
AVAILABLE: asb, jlink, svl
CURRENT: upload_protocol = svl
Looking for upload port...
Error: Please specify `upload_port` for environment or use global `--upload-port` option.
For some development platforms it can be a USB flash drive (i.e. /media/<user>/<device name>)
*** [upload] Explicit exit, status 1
======================================================================= [FAILED] Took 1.11 seconds =======================================================================
The terminal process "platformio 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

This is no big deal, I can see that the compilation went fine, but a bit surprising the first time - it goes against the convention that "build" only builds does not try to upload, and indicates fail when it should actually indicate success - since compilation was successful.

nigelb, 2021-11-22:
Hi @jerabaul29,
I agree it is odd. I am not 100% sure, but I think this used to be the default behavior in platformio because I had to figure out how to make it happen. But when I test on other platforms now it no longer auto uploads.

I will remove the auto-upload.

jerabaul29, 2021-11-23:
Sounds good :) .

jerabaul29, 2021-12-04:
This looks to be working now I think (did a fresh install a few minutes ago and I can do some compiles without uploads :) ). Feel free to close :) .

closed 2021-12-04 by nigelb