nigelb/platform-apollo3blue #31: very different amounts of memory used by i) Arduino IDE 1.8 vs ii) this platformio core

jerabaul29, 2021-12-04:
When compiling an empty sketch with Arduino IDE 1.8 using the core 1.2.3 I get the following memory use:

[image]

[image]

i.e., if I understand right, 8500 bytes of flash and 33820 bytes of RAM.

But, if I use the platformio core, using the v1.2.3 package, I get:

[image]

i.e., it uses 988 bytes of flash (really low, but I do not care, there are lots of flash, the issue is with RAM!), and 81924 bytes of RAM (this is a lot more, and a bit of a problem, I am going to get empty for RAM on one of my projects with this "over consumption :( ).

Any idea what the reason is? I installed everything following: https://github.com/jerabaul29/OpenMetBuoy-v2021a/blob/main/doc/setup_platformio_environment/Instructions.md .

jerabaul29, 2021-12-04:
Mmh, I think that this high memory use looks like what I would expect with the RTOS, not the barebone. Weird, I thought I had followed the correct steps for a "barebone core installation". Seems like some older core v1 installs were giving right memory, see for example use: https://github.com/nigelb/platform-apollo3blue/issues/28 .

jerabaul29, 2021-12-04:
Mmmh, actually looks like the issue #28 screenshots using less RAM were actually with core v2, which is really strange; the core v2 should actually use a lot more RAM than the bare metal core v1.

nigelb, 2021-12-04:
Hi @jerabaul29,

I think this may be a bug that I fixed yesterday but haven't pushed.
When I fixed #15, the fix did not make it into the core v1 builder.
So it may not be reporting the correct sizes.

I just ran a test now with the following `platformio.ini`:
```ini
[env:SparkFun_RedBoard_Artemis_ATP]
platform = apollo3blue
board = SparkFun_RedBoard_Artemis_ATP
framework = arduino
platform_packages = framework-arduinoapollo3@1.2.3
build_type=release
```

and `src/main.cpp`:

```cpp
void setup(){}
void loop(){}
```

with the output:

```
.
.
.
arm-none-eabi-objcopy -O binary .pio/build/SparkFun_RedBoard_Artemis_ATP/program .pio/build/SparkFun_RedBoard_Artemis_ATP/firmware.bin
MethodWrapper(["checkprogsize"], [".pio/build/SparkFun_RedBoard_Artemis_ATP/program"])
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   0.0% (used 4 bytes from 393216 bytes)
Flash: [          ]   0.1% (used 988 bytes from 983040 bytes)
.pio/build/SparkFun_RedBoard_Artemis_ATP/program  :
section             size        addr
.text                988       65536
.stack             81920   268435456
.data                  0   268517376
.bss                   4   268517376
.ARM.attributes       52           0
.debug_info       119050           0
.debug_abbrev       9336           0
.debug_aranges       936           0
.debug_ranges       1136           0
.debug_line        19634           0
.debug_str         37853           0
.comment             102           0
.debug_frame        2096           0
.debug_loc          8362           0
Total             281469
```

However, when I disable the fix I get:

```
.
.
.
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  20.8% (used 81924 bytes from 393216 bytes)
Flash: [          ]   0.1% (used 988 bytes from 983040 bytes)
text	   data	    bss	    dec	    hex	filename
    988	      0	  81924	  82912	  143e0	.pio/build/SparkFun_RedBoard_Artemis_ATP/program
```

Which matched what you are seeing.
I want to finish some more tests before I push these changes, but is won't take too long.

jerabaul29, 2021-12-04:
Ok, many thanks for the fast answer :) .

nigelb, 2021-12-05:
Hi @jerabaul29,

I have pushed the update.
Give it a go and let me know how you go.

jerabaul29, 2021-12-05:
Yes, pulled main now, all looks good, the memory is reported as 4 bytes of RAM and 988 bytes of Flash, as expected :) . Feel free to close :) . Many thanks again :) .

closed 2021-12-06 by nigelb