nigelb/platform-apollo3blue #32: caching of compilation?

jerabaul29, 2021-12-07:
I am ("as usual" :) ) using the core v1, following https://github.com/jerabaul29/OpenMetBuoy-v2021a/blob/main/doc/setup_platformio_environment/Instructions.md (which should be similar to your instructions).

I was curious about the builds caching: what is cached, where, when? It looks like each time I re-open the PlatformIO project, the whole thing gets recompiled, even if nothing changed since last time my computer was switched on off. May it be possible to cache to a disk folder in a "." folder in the tree of the project to avoid slow compilation at project re opening?

I was wondering: what kind of caching algorithm is use otherwise, for example if only a couple of files are modified? :)

nigelb, 2021-12-08:
Hi @jerabaul29,

My understanding is that the compilation caching is all handled by the core of PlatformIO.
I did not add anything to this repository in regards to compilation caching.

However, I expect that a full rebuild when you open your project is directed by your IDE as I don't recall this happening when using the cli tools. The only time I have experienced PlatformIO do a full rebuild by itself is when I modify the `platformio.ini` file, or run `pio run -t clean`. There are probably a few other instances where this could happen, like maybe when something is upgraded, but I am speculating and haven't tested this to know for sure.

jerabaul29, 2021-12-08:
Interesting. Thanks for the explanation. Then I will monitor this a bit and report to platformio-core if needed. I think you may be right, that platformio tries to update stuff "in the shadows" each time it starts, and that if some files were changed by these updates "in the shadow", that may be what causes a full re build: I had a full re-build yesterday when starting my computer, but not today...

closed 2021-12-08 by jerabaul29