nigelb/platform-apollo3blue #33: unity unit test framework not recognized

jerabaul29, 2021-12-07:
I am trying to set up some unit tests following:

https://github.com/platformio/platformio-examples/blob/develop/unit-testing/calculator/test/test_embedded/test_calculator.cpp

But the unity package (likely what provides the unit testing functionalities) does not seem to be recognized:

[image]

Any idea why / how to fix this? :) I am on the core v1 as usual :) .

nigelb, 2021-12-08:
Hi @jerabaul29,

### tldr;
I believe there are two issues at play. 
Firstly, PlatformIO automatically includes the unity framework in the include path _only_ when compiling the source files in the test directory. And the second is that unless you run have run the tests, you may not have the `tool-unity` package installed.
The other thing I will note is that the core v2 behaves differently than expected because the unity test framework seem to already be included in the source files. 

With both of these sorted, you can see it works fine for me:
[image]

To run the tests from vscode, in the PlatformIO Tab, under Project Tasks > Advanced select the Test option:
[image]

### Long Version

Here I demonstrate that an Apollo3 Core V1 project behaves the same as an Arduino Mega 2560 project with regards to the inclusion of the unity test framework.
 
Create a project:
```
$ mkdir test_project
$ cd test_project
$ pio init -b SparkFun_RedBoard_Artemis_ATP --ide vscode -O"platform_packages=framework-arduinoapollo3@1.2.3"

The current working directory /home/user/test_project will be used for the project.

The next files/directories have been created in /home/user/test_project
include - Put project header files here
lib - Put here project specific (private) libraries
src - Put project source files here
platformio.ini - Project Configuration File

Project has been successfully initialized including configuration files for `vscode` IDE.
```
Create a file in `src/main.cpp` with the contents:
```c++
#include "Arduino.h"
#include <unity.h>

void setup()
{
    Serial.begin(115200);

}

void loop()
{
    Serial.println("Hello World!");
    delay(1000);
}
```
Create a file in `test/test_1.cpp` with the following contents:

```c++
#include "Arduino.h"
#include <unity.h>

void test_flash_init(void)
{
    TEST_ASSERT_TRUE(true);
}

void setup()
{
    Serial.begin(115200);
    UNITY_BEGIN();

    RUN_TEST(test_flash_init);
    UNITY_END();
}

[... 255 lines cut ...]
Compiling .pio/build/SparkFun_RedBoard_Artemis_ATP/core-implement/CommonPulse.cpp.o
Indexing .pio/build/SparkFun_RedBoard_Artemis_ATP/libvariant.a
Archiving .pio/build/SparkFun_RedBoard_Artemis_ATP/libmbed_bridge.a
Indexing .pio/build/SparkFun_RedBoard_Artemis_ATP/libmbed_bridge.a
Archiving .pio/build/SparkFun_RedBoard_Artemis_ATP/libcore-implement.a
Indexing .pio/build/SparkFun_RedBoard_Artemis_ATP/libcore-implement.a
Linking .pio/build/SparkFun_RedBoard_Artemis_ATP/program
Checking size .pio/build/SparkFun_RedBoard_Artemis_ATP/program
arm-none-eabi-objcopy -O binary .pio/build/SparkFun_RedBoard_Artemis_ATP/program .pio/build/SparkFun_RedBoard_Artemis_ATP/firmware.bin
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   7.5% (used 29384 bytes from 393216 bytes)
Flash: [=         ]  12.0% (used 117540 bytes from 983040 bytes)
====================================================================================== [SUCCESS] Took 2.71 seconds ======================================================================================
```

The bloody thing compiles fine.
This is because the core v2 framework has the unity file already in it:

```
$ cd ~/.platformio/packages/framework-arduinoapollo3@2.1.0/
$ find . | grep unity.h
./cores/mbed-os/features/frameworks/unity/unity/unity.h
./cores/mbed-os/features/frameworks/utest/source/unity_handler.cpp
./cores/mbed-os/features/frameworks/utest/utest/unity_handler.h
```

jerabaul29, 2021-12-08:
Many thanks for the in-depth debugging :) . Definitely something strange going on.

Interesting :) . Actually it looks like this has gotten solved "all by itself", I cannot see the squiggles any longer.

A few points regarding the error I had:

- I do not import unity in any file in my "src folder code", only in my "test folder code", so I think I am doing this correctly :) . The screenshot was maybe not so clear, but was I showed actually lived in the "test folder" (this is visible from the initial error message, that names that this lives in a sub folder of the test folder).
- I am able to compile both the "src code" and the "test folder", so this means that, when building stuff, all is fine, and the needed unity stuff is well present on my machine (confirmed by checking that ```~/.platformio/packages/tool-unity``` does exist and is well on my machine) - probably an additional confirmation that I am setting stuff right.

The strange thing is that, this morning, the squiggles are gone...

I wonder if unity was not installed by default by PlatformIO, and got installed only the first time I run the command, or if the autodetect of the packages was re-run when I re-opened PlatformIO this morning, or something like that... That could explain why I had squiggles initially, and now they are not here any longer.

Sorry for the noise, but definitely something a bit weird going on... May be well possible that this is only a "platformio warmup issue that requests to run tests once, and re load, to be enabled". Weird. Wonder if this is maybe a platformio-core issue then, in which case maybe we should report it? :) What do you think? :) .

jerabaul29, 2021-12-08:
Ok, I did some tests on another computer, and it does look like this is some "IntelliSense warmup issues". I think that there may be a need to run a test one and then restart platformio or something like this for getting all up and going. I think the confusion may come from these differences between core v1 and core v2, the warmup etc.

So after some warmup etc all seems to be fine... Thanks for the help and the discussion, sorry for the noise!

closed 2021-12-08 by jerabaul29