nigelb/platform-apollo3blue #34: brainstorming: how would you recommend to compile / "install" an external non arduino library

jerabaul29, 2022-01-28:
I am asking here, as I feel that having a platformio core set up may actually make things quite a bit easier (ie the right compiler options etc are already in place), but I may be wrong; if this is off topic, just let me know :) .

I would like to use this library in a project: https://github.com/drowe67/codec2 . However, it does not come as an arduino library, but as a standard C++ library with CMake. There is a way to compile it for ARM microcontrollers it seems though:

> On ARM machines, setting the C Flag __EMBEDDED__ and linking with the ARM CMSIS library will improve performance on ARM-based microcontrollers. __REAL__ and FDV_ARM_MATH are additional ARM-specific options that can be set to improve performance if required, especially with OFDM modes.

Any idea if I can take "advantage" of the platformio core to help me compile this / how I could do so? :)

nigelb, 2022-02-04:
Hi @jerabaul29,

You could try checking out the library in your `project_dir/lib` directory and then create a `library.json` file (in the checked out directory) as described [here](https://docs.platformio.org/en/latest/librarymanager/config.html), an [example](https://github.com/jcu-eresearch/Arduino-BLEUart/blob/main/library.json).

Then fiddle with the settings and see how you go. If I get some time I will see if I can get it to compile.

jerabaul29, 2022-02-04:
Ok, thanks, will have a try some day soon :) . This is very kind of you to provide some pointers, don t feel forced to do any work on your side of course :) .

nigelb, 2022-03-31:
Hi @jerabaul29,

How did you go with this? I had a play.

If you run these commands to create a new project:

```bash
mkdir code2_test
cd code2_test
pio init --board SparkFun_RedBoard_Artemis_ATP
mkdir scripts
cd lib
mkdir Trace
mkdir Trace/src
git clone https://github.com/drowe67/codec2.git
```
Then create `Trace/src/Trace.h`
```c++
#ifndef TRACE_H
#define TRACE_H

#include "stdlib.h"

size_t trace_write(char* buffer, size_t buf_idx_ptr);

#endif
```
and create `Trace/library.json`:
```json
{
    "name": "Trace",
    "version": "0.0.1",
    "description": "",
    "repository": {
      "type": "git",
      "url": ""
    },
    "keywords": [
      
    ],
    "authors": [
      {
        "name": ""
      }
    ],
    "frameworks": [
      "arduino"
    ],
    "dependencies": {
        "Trace": ">=0.01"
    },
    "license": "",
    "libArchive": false,
    "build": {
      "srcFilter": [
        "+<*.c>",
        "+<*.h>"
      ]
    }
  }
```

Then create `codec2/library.json`
```json
{
    "name": "Codec 2",
    "version": "1.0.3",
    "description": "Codec 2 is an open source (LGPL 2.1) low bit rate speech codec: http://rowetel.com/codec2.html",
    "repository": {
      "type": "git",
[... 285 lines cut ...]
Compiling .pio/build/SparkFun_RedBoard_Artemis_ATP/mbed_bridge/core-implement/CommonInterrupt.cpp.o
Compiling .pio/build/SparkFun_RedBoard_Artemis_ATP/mbed_bridge/core-implement/CommonMath.cpp.o
Compiling .pio/build/SparkFun_RedBoard_Artemis_ATP/mbed_bridge/core-implement/CommonPulse.cpp.o
Compiling .pio/build/SparkFun_RedBoard_Artemis_ATP/mbed_bridge/core-implement/CommonTiming.cpp.o
Compiling .pio/build/SparkFun_RedBoard_Artemis_ATP/mbed_bridge/core-implement/HardwareSerial.cpp.o
Compiling .pio/build/SparkFun_RedBoard_Artemis_ATP/mbed_bridge/core-implement/Yield.cpp.o
Compiling .pio/build/SparkFun_RedBoard_Artemis_ATP/mbed_bridge/core-implement/itoa.c.o
Compiling .pio/build/SparkFun_RedBoard_Artemis_ATP/mbed_bridge/main.cpp.o
Compiling .pio/build/SparkFun_RedBoard_Artemis_ATP/core-implement/CommonAnalog.cpp.o
Compiling .pio/build/SparkFun_RedBoard_Artemis_ATP/core-implement/CommonPulse.cpp.o
Archiving .pio/build/SparkFun_RedBoard_Artemis_ATP/libmbed_bridge.a
Indexing .pio/build/SparkFun_RedBoard_Artemis_ATP/libmbed_bridge.a
Archiving .pio/build/SparkFun_RedBoard_Artemis_ATP/libcore-implement.a
Indexing .pio/build/SparkFun_RedBoard_Artemis_ATP/libcore-implement.a
Linking .pio/build/SparkFun_RedBoard_Artemis_ATP/program
Checking size .pio/build/SparkFun_RedBoard_Artemis_ATP/program
arm-none-eabi-objcopy -O binary .pio/build/SparkFun_RedBoard_Artemis_ATP/program .pio/build/SparkFun_RedBoard_Artemis_ATP/firmware.bin
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [======    ]  60.5% (used 237984 bytes from 393216 bytes)
Flash: [==        ]  20.5% (used 201264 bytes from 983040 bytes)
============================================================================= [SUCCESS] Took 14.97 seconds =============================================================================
```

Now you may have some more issues when you actually start to use the library.
The unimplemented function in `lib/Trace/src/Trace.h` comes to mind.

jerabaul29, 2022-03-31:
Sorry, I got a big bunch of work and I have not had the time to look at this myself :( . Unsure of when I will have time. But many thanks, this indeed looks very good. Is it ok to either keep this issue open, or move it to a discussion, so that I can easily find it back when I come back to this? :) .

nigelb, 2022-04-08:
Hi @jerabaul29,

I will convert this Issue to a discussion under Q & A.