nigelb/platform-apollo3blue #39: Error building a blink application with artemis nano

tgruetzm, 2022-06-09:
With any builds for the artemis nano board I get the following error:
arm-none-eabi-g++: error: blink\.pio\build\SparkFun_RedBoard_Artemis_Nano\program.map: No such file or directory
*** [.pio\build\SparkFun_RedBoard_Artemis_Nano\program.exe] Error 1

Any thoughts on what is causing it?

nigelb, 2022-06-10:
Hi @tgruetzm,

I do not see your problem when I try. These are the steps I took to try and replicate your issue:

Install  `platform-apollo3blue`
```bash
$ pio platform install git+https://github.com/nigelb/platform-apollo3blue
*******************************************************************************************************************************************************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for embedded development > https://platformio.org/platformio-ide
*******************************************************************************************************************************************************************************************************************

WARNING: This command is deprecated and will be removed in the next releases. 
Please use `pio pkg install` instead.

Platform Manager: Installing git+https://github.com/nigelb/platform-apollo3blue
git version 2.25.1
Cloning into '/home/pio/.platformio/.cache/tmp/pkg-installing-ka_w20p9'...
remote: Enumerating objects: 79, done.
remote: Counting objects: 100% (79/79), done.
remote: Compressing objects: 100% (65/65), done.
remote: Total 79 (delta 12), reused 46 (delta 4), pack-reused 0
Unpacking objects: 100% (79/79), 907.88 KiB | 4.66 MiB/s, done.
Platform Manager: apollo3blue@0.0.2+sha.4a3a7fb has been installed!
Tool Manager: Installing platformio/toolchain-gccarmnoneeabi @ 1.90201.191206
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: toolchain-gccarmnoneeabi@1.90201.191206 has been installed!
The platform 'git+https://github.com/nigelb/platform-apollo3blue' has been successfully installed!
The rest of the packages will be installed later depending on your build environment.

```

Initialize project:
```bash
$ pio init -b SparkFun_RedBoard_Artemis_Nano -O"platform_packages=framework-arduinoapollo3@https://github.com/sparkfun/Arduino_Apollo3#v2.2.0"

The current working directory /home/pio/projects/ucontroller/Apollo3/testing/core_v2 will be used for the project.

The next files/directories have been created in /home/pio/projects/ucontroller/Apollo3/testing/core_v2
include - Put project header files here
lib - Put here project specific (private) libraries
src - Put project source files here
platformio.ini - Project Configuration File
Resolving SparkFun_RedBoard_Artemis_Nano environment packages...
Tool Manager: Installing git+https://github.com/sparkfun/Arduino_Apollo3#v2.2.0
git version 2.25.1
Cloning into '/home/pio/.platformio/.cache/tmp/pkg-installing-pnrfevq9'...
remote: Enumerating objects: 279, done.
remote: Counting objects: 100% (279/279), done.
remote: Compressing objects: 100% (229/229), done.
remote: Total 279 (delta 82), reused 151 (delta 36), pack-reused 0
Receiving objects: 100% (279/279), 2.62 MiB | 4.55 MiB/s, done.
Resolving deltas: 100% (82/82), done.
Note: switching to '771ce8fe14b54425775bb48c777709bf28eb9100'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -
[... 230 lines cut ...]
	Sending frame #54, length: 2048
	Sending frame #55, length: 2048
	Sending frame #56, length: 2048
	Sending frame #57, length: 2048
	Sending frame #58, length: 2048
	Sending frame #59, length: 2048
	Sending frame #60, length: 2024

	Upload complete

	Nominal bootload bps: 46404.72
```

Run program:
```
�Startup....
Loop
Loop
Loop
Loop
Loop
Loop
```

Which is working fine. Do you see anything different when you try?

tgruetzm, 2022-06-11:
I'm not sure what happened, but I removed and reinstalled everything and it seems to work now if I create new projects.  My old projects still get the same error though.  I guess I'll just re-create them.  Thanks for looking!

tgruetzm, 2022-06-11:
Hey @nigelb,  I was still getting this issue and after some additional testing it seems like this only occurs when my project name has spaces.  When I used "test" as my project it worked fine, then I re-created my project and it still didn't work.  Then I added underscores and it works.  Maybe you could see if you can reproduce it that way?

nigelb, 2022-06-15:
Hey @tgruetzm,
What operating system are you using?
I could not replicate your issue on Linux but I could when I went and tried on Windows.
I cannot test on a mac.

Now that I can replicate the issue, I can fix it.

tgruetzm, 2022-06-16:
@nigelb, yes I'm using windows.  I was thinking it may be is specific.

tgruetzm, 2022-07-07:
Hey @nigelb 

What would I need to do to test/how to I get your change to run with my PIO?  I tried running pio platform install git+https://github.com/nigelb/platform-apollo3blue but I don't think that installed anything new.

nigelb, 2022-07-08:
Hey @tgruetzm,

From my project directory, this worked for me:

```bash
$ pio pkg update
Resolving SparkFun_RedBoard_Artemis_Nano environment packages...
Platform Manager: Updating apollo3blue @ 0.0.2+sha.4a3a7fb
git version 2.25.1
remote: Enumerating objects: 18, done.
remote: Counting objects: 100% (18/18), done.
remote: Compressing objects: 100% (5/5), done.
remote: Total 11 (delta 6), reused 11 (delta 6), pack-reused 0
Unpacking objects: 100% (11/11), 987 bytes | 141.00 KiB/s, done.
From https://github.com/nigelb/platform-apollo3blue
   4a3a7fb..a4fe6a5  master     -> origin/master
Updating 4a3a7fb..a4fe6a5
Fast-forward
 builder/frameworks/arduino/core_v2.py | 2 +-
 builder/helpers/uploader.py           | 4 ++--
 2 files changed, 3 insertions(+), 3 deletions(-)
Already up-to-date.
```

commit 5a7c45aa (2022-07-11): Fix #39 in core_v1 as well.