nigelb/platform-apollo3blue #42: new warning: "No baud rate specified. Using default svl baud rate of 921600"

jerabaul29, 2022-12-14:
I updated the present platformio core (many thanks for your awesome continued work on this :) ) and I  recompiled an old project, now getting the warning:

```
No baud rate specified.
Using default svl baud rate of 921600
```

This is not a big issue, everything compiles fine etc, and I am happy with using this svl baudrate but wonder if there is a way to get it to go away, just "for the sake of purity"? :) . I guess this is the sparkfun variable loader that uploads the code to the board complaining its baudrate to use is not set. Not sure why it was not here before / I had not noticed it before.

nigelb, 2022-12-15:
Hi @jerabaul29,

I am glad you are finding this project useful!

This warning is not that new: d75d5e8fe34bc9e9158f98c4fed216ab8ac4fb50, however it will now only show up in verbose mode.

closed 2022-12-15 by nigelb

jerabaul29, 2022-12-15:
Thanks :) .

A small note: wonder if a more descriptive error message could be useful? Something like:

```
sys.stderr.write("No baud rate specified; please specify parameter, example "upload_speed = 921600", in your platformio.ini file.\n")
```

maybe? This makes it a bit clearer what the missing parameter name is, and where it could be added?

jerabaul29, 2022-12-16:
Amazing, many thanks :) .

commit 8e9d57b9 (2022-12-15): Fix #42, warnings now only show in verbose mode.