nigelb/platform-apollo3blue #43: Cannot build for boards that are not in boards-sfe

fariouche, 2022-12-20:
Hello,

This is a followup of the platformio discussion so that we do not forget.
I had to modify one line in ambiqsuitesdk-sfe.py to make it work:

libs.append(env.BuildLibrary(
    join("$BUILD_DIR", "Variant"),
    join(FRAMEWORK_DIR, *framework_build.get("variant"), "bsp"),
))

Look at the added “bsp” at the end.
The reason is that my board’s bsp is “variant”: [“boards”, “apollo3_evb”]
and inside the boards/apollo3_evb are two folders, bsp and examples… and platformio will try to build the examples here!
The folders inside boards-sfe only contains bsp, so the problem is not visible

This modification is backward compatible.

This is the only patch I still need. (I do have one more, to have stack traces when HardFault is called, but I can override the symbol from my application. If you want me to send you this patch too, let me know and I will create an other issue)
Thanks

fariouche, 2022-12-22:
diff --git a/builder/frameworks/ambiqsuitesdk-sfe.py b/builder/frameworks/ambiqsuitesdk-sfe.py
index 1c0cad5..3f0eacb 100644
--- a/builder/frameworks/ambiqsuitesdk-sfe.py
+++ b/builder/frameworks/ambiqsuitesdk-sfe.py
@@ -278,7 +278,7 @@ libs = []
 
 libs.append(env.BuildLibrary(
     join("$BUILD_DIR", "Variant"),
-    join(FRAMEWORK_DIR, *framework_build.get("variant")),
+    join(FRAMEWORK_DIR, *framework_build.get("variant"), "bsp"),
 ))
 
 libs.append(env.BuildLibrary(

nigelb, 2022-12-22:
Hi @fariouche,

I have added a new board `Ambiq_Apollo3_Blue_EVB` and stopped it from compiling the examples directory.

Give it a try and let me know how you go.

fariouche, 2022-12-22:
Hi @nigelb 
Thank you, this is even better! It seems like this will enable the possibility to compile apollo4 boards too
I have my own json board file. I will adapt it then and let you know

fariouche, 2022-12-22:
ok, almost good ;)
I had to modify the src filter like this:
"variant_lib_src_filter": "+<bsp/*>"

Else, it is not compiling what is inside the bsp folder, missing symbols like am_bsp_low_power_init

nigelb, 2022-12-22:
Ok I have added that too.

fariouche, 2022-12-22:
all good! Thank you very much!

closed 2022-12-22 by fariouche