nigelb/platform-apollo3blue #45: Some dependency issues with the libraries

fariouche, 2022-12-23:
Hello,

I was trying to integrate the amota example... for that I had to include the dependencies "ambiq_ble_apps_amota", "ble-host" and "ble-profiles"
Doing this immediately triggered a huge display of dependencies from platformio.... for 2 minutes.
I see this kind of output when there is a circular dependency... here the 3 libraries above depends on each others.

I then tried to disable LDF mode.... and encountered missing dependencies in (at least) the following libraries:
uECC, ble-host, ble-profiles.

I've used the following deps to solve the problem:
```
diff --git a/extra/ambiqsdk-sfe/libraries/third_party/cordio/ble-host/library.json b/extra/ambiqsdk-sfe/libraries/third_
party/cordio/ble-host/library.json
index 396400e..b103b12 100644
--- a/extra/ambiqsdk-sfe/libraries/third_party/cordio/ble-host/library.json
+++ b/extra/ambiqsdk-sfe/libraries/third_party/cordio/ble-host/library.json
@@ -8,7 +8,8 @@
   "headers": ["hci_handler.h"],
   "license": "APACHE-2.0",
   "dependencies": {
-    "uECC":"^0.1"
+    "uECC":"^0.1",
+    "ble-profiles": "*"
   },
   "build": {
     "flags": [
@@ -42,4 +43,4 @@
       "-<sec/pal/sec_ccm.c>"
     ]
   }
-}
\ No newline at end of file
+}
```
```
diff --git a/extra/ambiqsdk-sfe/libraries/third_party/cordio/ble-profiles/library.json b/extra/ambiqsdk-sfe/libraries/third_party/cordio/ble-profiles/library.json
index 0e9b248..6797aa6 100644
--- a/extra/ambiqsdk-sfe/libraries/third_party/cordio/ble-profiles/library.json
+++ b/extra/ambiqsdk-sfe/libraries/third_party/cordio/ble-profiles/library.json
@@ -7,6 +7,10 @@
   ],
   "headers": ["app_api.h"],
   "license": "APACHE-2.0",
+  "dependencies": {
+    "ble-host": "*",
+    "cordio-wsf": "*"
+  },
   "build": {
     "flags": [
       "-I%(FRAMEWORK_DIR)s/third_party/cordio/ble-profiles",
@@ -79,4 +83,4 @@
       "+<*.h>"
     ]
   }
-}
\ No newline at end of file
+}
```
```
diff --git a/extra/ambiqsdk-sfe/libraries/third_party/uECC/library.json b/extra/ambiqsdk-sfe/libraries/third_party/uECC/library.json
index f20e41e..15eb913 100644
--- a/extra/ambiqsdk-sfe/libraries/third_party/uECC/library.json
+++ b/extra/ambiqsdk-sfe/libraries/third_party/uECC/library.json
@@ -6,6 +6,11 @@
         "ambiqsdk-sfe"
     ],
     "license": "BSD 2-clause",
+    "dependencies": {
+        "uECC":"^0.1",
+        "ble-profiles": "*",
+        "cordio-wsf": "*"
+    },
     "build": {
         "flags": [],
         "libArchive": false,
@@ -17,4 +22,4 @@
             "-<uECC_ll.c>"
         ]
     }
-}
\ No newline at end of file
+}
```

The build is so much faster without the LDF from platformio !!! :+1:

fariouche, 2023-01-24:
Hello @nigelb , just wanted to let you know that I have some working great, thank you!

Just, if you can push the modifications of the dependencies above, it will be perfect, so that I don't have to fork!

By the way, I'm using the ambiq suite sdk 3.0.0 (official from ambiq web site), with no modifications, and it works great!

nigelb, 2023-02-02:
Hi @fariouche,

Sorry I haven't got back to you sooner.
I injured my self on holiday and it is making it very difficult to do much.
Good news about ambiq suite sdk 3.0.0, that is something I had planned to test out.

I have pushed these changes. Have a go and let me know.

fariouche, 2023-02-11:
Thank you so much @nigelb , everything is working as expected! :heart:

closed 2023-02-11 by fariouche

nigelb, 2023-02-14:
Hi @fariouche, that is good news. Thanks!