nigelb/platform-apollo3blue #46: bug: strange include failure for /lib code

jerabaul29, 2023-01-23:
Forwarding from https://github.com/platformio/platformio-core/issues/4513# following @ivankravets recommendation.

Copy of the issue:

I am hitting a strange include failure for some /lib code in a PlatformIO project (I put nearly all the code in /lib these days because of some habits I grabbed from some projects where I run quite a few unit-tests, which can include code from /lib but not from /src ; if putting most of my code in /lib is an antipattern, sorry, let me know).

---

To reproduce the exact same setup:

- install the sparkfun artemis core in v 1.2.3: https://github.com/nigelb/platform-apollo3blue , detailed instructions https://github.com/jerabaul29/OpenMetBuoy-v2021a/blob/main/development_environment/setup_platformio_environment/Instructions.md .

- set up the repo:

```
git clone git@github.com:jvoermans/Wind_Input_V1.git
git checkout logger_version_jr-import_issue_sdfat
git submodule init
git submodule update
```

- try to compile in PlatformIO; I get the log and error:

```
Executing task: platformio run 

Processing SparkFun_RedBoard_Artemis (platform: apollo3blue; board: SparkFun_RedBoard_Artemis; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/apollo3blue/SparkFun_RedBoard_Artemis.html
PLATFORM: Apollo 3 Blue (0.0.2) > SparkFun RedBoard Artemis
HARDWARE: AMA3B1KK 48MHz, 384KB RAM, 960KB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES: 
 - framework-arduinoapollo3 @ 1.2.3 
 - toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 26 compatible libraries
Scanning dependencies...
Dependency Graph
|-- data_manager
|-- config
|   |-- SdFat @ 2.2.0
|   |   |-- SPI @ 1.0
|   |-- utils
|-- gnss_manager
|   |-- Embedded Template Library - Arduino @ 20.35.5
|   |-- config
|   |   |-- SdFat @ 2.2.0
|   |   |   |-- SPI @ 1.0
|   |   |-- utils
|   |-- kiss_clang_posix_time_utils
|   |-- statistical_processing
|   |   |-- Embedded Template Library - Arduino @ 20.35.5
|   |   |-- config
|   |   |   |-- SdFat @ 2.2.0
|   |   |   |   |-- SPI @ 1.0
|   |   |   |-- utils
|   |-- time_manager
|   |   |-- config
|   |   |   |-- SdFat @ 2.2.0
|   |   |   |   |-- SPI @ 1.0
|   |   |   |-- utils
|   |   |-- kiss_clang_posix_time_utils
|   |   |-- utils
|   |-- watchdog_manager
|   |   |-- WDT @ 0.1
|   |-- Adafruit GPS Library @ 1.7.1
|   |   |-- SPI @ 1.0
[... 125 lines cut ...]
    4 | #include "SdFat.h"
      |          ^~~~~~~~~
compilation terminated.
*** [.pio/build/SparkFun_RedBoard_Artemis/lib2db/utils/print_utils.cpp.o] Error 1
==================================================================================================== [FAILED] Took 14.62 seconds ====================================================================================================

 *  The terminal process "platformio 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

---

Now what I really cannot understand is that the error is that seems to break compilation is:

```
lib/config/firmware_configuration.h:4:10: fatal error: SdFat.h: No such file or directory
```

But the ```SdFat.h``` file does is present, in its own lib folder as a submodule, is recognized by the dependency checker built in PlatformIO, etc. Actually, just uncommenting the line:

https://github.com/jvoermans/Wind_Input_V1/blob/f7e8ae031db1b96a731e147fc247a85a0be974e0/lib/utils/print_utils.h#L7

ie importing SdFat.h directly in the print_utils.h header (note that SdFat.h should already be available in print_utils.h due to importing firmware_configuration.h that itself imports SdFat.h) fixes things, which is weird, as it shows that the compiler actually then knows of where to find SdFat.h

Any idea why the compiler cannot find SdFat.h when trying to build print_utils.cpp in a case and not in the other?

nigelb, 2023-02-02:
Hi @jerabaul29,

If you run platformio in verbose mode (with the -v flag) you can see the command line that is passed to the compiler.
I normally compile a few times without cleaning so that there is less noise in the output and then do a verbose compile.
With this you can verify that all of the include paths have been passed to the compiler.

What OS are you compiling on? Does this issue persist when you compile on another OS?

I ask this because last year I ran into a weird `#include` bug where including one of my project headers would cause the Arduino core compile to fail with all kinds of strange errors, but only on Windows.
I moved the include to another file and the error went away.  Very strange.

nigelb, 2023-05-25:
Hi @jerabaul29,

Just wondering how you went with this?
If you have a small example project, with the bug, that I can play with I may have some luck narrowing down the issue.

Nigel