nigelb/platform-apollo3blue #47: Make BOARD_VARIANTS_DIR customizable

caveman99, 2023-06-30:
When defining custom board json files in your own project (short of adding a new core to the framework proper) the BOARD_VARIANTS_DIR points to the framework dir looking for the flag files in the wrong place. With other processor architectures i can do the following:

```
[env:rak11720]
extends = apollo3_base
board = wiscore_rak11720
board_level = extra
build_flags = ${apollo3_base.build_flags} -Ivariants/rak11720
```

and put the respective JSON in the boards subdir of my project. If i do that with this core, it tries to find the support files in `.platformio\packages\framework-arduinoapollo3\variants\WisCore_RAK11720_Board\mbed` instead of variants\rak11720\mbed in the project dir. Ideally i wanna specify the project specific include path in the custom JSON

nigelb, 2023-07-01:
Hi @caveman99, 

Have a go with this update and let me know how you go.
If you have a directory `$PROJECT_DIR/variants` it will now look in there.
Keep in mind that in your example it will actually look in: `$PROJECT_DIR/variants/WisCore_RAK11720_Board`

Nigel.

caveman99, 2023-07-01:
working fine, thanks. Since i also control the json file and i want the directory structure to be somewhat similar to other platforms, i changed the variant name in the JSON to match the dir name under variants. If the platform is working for our project one day i intend to contribute the wisblock variant definition here.

closed 2023-07-01 by caveman99