nigelb/platform-apollo3blue #48: Build fails in PlatformIO

Leenix, 2023-07-24:
As title, building a project fails with the following error message:

```
TypeError: expected str, bytes or os.PathLike object, not NoneType:
  File "/home/karl/.platformio/penv/lib/python3.10/site-packages/platformio/builder/main.py", line 175:
    env.SConscript("$BUILD_SCRIPT")
  File "/home/karl/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/karl/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/karl/.platformio/platforms/apollo3blue/builder/main.py", line 125:
    env.SConscript(join("helpers", "targets.py"), exports="env")
  File "/home/karl/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/karl/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/karl/.platformio/platforms/apollo3blue/builder/helpers/targets.py", line 177:
    add_ota_image(env)
  File "/home/karl/.platformio/platforms/apollo3blue/builder/helpers/targets.py", line 108:
    APOLLO3_SCRIPTS_DIR=join(FRAMEWORK_DIR, "tools", "apollo3_scripts"),
  File "/usr/lib/python3.10/posixpath.py", line 76:
    a = os.fspath(a)
```

Leenix, 2023-07-24:
Build fixed by changing the following line in `apollo3blue>builder>helpers>targets.py>add_ota_image`:

`FRAMEWORK_DIR = platform_apollo3blue.get_package_dir("framework-ambiqsuitesdkapollo3-sfe")`

changed to...

` FRAMEWORK_DIR = platform_apollo3blue.get_package_dir("framework-arduinoapollo3")`

nigelb, 2023-07-28:
Hey @Leenix,

Is this still an issue? I accidentally pushed to the wrong branch.
But I thought I had also pushed a fix.

Nigel.

nigelb, 2023-07-28:
Hmm...

Actually I think I see the issue.

I will fix it and let you know.

Nigel.

nigelb, 2023-07-28:
Hey @Leenix,

I think that should have it.
Have a go and let me know.

Nigel.

Leenix, 2023-08-14:
Looks to be fixed.

closed 2023-08-14 by Leenix