nigelb/platform-apollo3blue #50: How to set -Os instead of default -O0?

fbenamrouche, 2023-12-07:
Hello,

How can I override the default -O0, when using ambiqsuite-sdk?
i suppose this requires a modification of the builder?

Thanks!

nigelb, 2023-12-13:
Hi @fbenamrouche,

You can try adding a `build_flags` to your `platformio.ini` file:

```ini
build_flags=-Os
```

If the builder needs to be modified I can probably make that happen on the weekend.

Nigel.

fbenamrouche, 2023-12-13:
The -O0 is hardcoded in the builder.
But I think using build_unflags=-O0 works together with build_flags=-0s

nigelb, 2024-01-10:
Hi @fbenamrouche,

Sorry for taking so long to get back to this.
I have been looking into this and when I look at [GCC's](https://linux.die.net/man/1/gcc) man page I see this:

> If you use multiple -O options, with or without level numbers, the last such option is the one that is effective. 

And the `platformio.ini`'s `build_flags` end up after the build flags set by the platform:

```
arm-none-eabi-gcc -o .pio/build/SparkFun_RedBoard_Artemis/libcc3/wsf/port/freertos/wsf_assert.o -c -mthumb -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -ffunction-sections -fdata-sections -fomit-frame-pointer -MMD -MP -std=c99 -Wall -g -O0 -Werror=return-type -Wall -Os ....
```

And when I tested this I definitely get a different build size:

With `-Os` in `build_flags`:
```
arm-none-eabi-objcopy -O binary .pio/build/SparkFun_RedBoard_Artemis/program .pio/build/SparkFun_RedBoard_Artemis/firmware.bin
Checking size .pio/build/SparkFun_RedBoard_Artemis/program
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  10.8% (used 42568 bytes from 393216 bytes)
Flash: [=         ]  11.4% (used 112228 bytes from 983040 bytes)
```

Without `-Os` in `build_flags`:
```
Checking size .pio/build/SparkFun_RedBoard_Artemis/program
arm-none-eabi-objcopy -O binary .pio/build/SparkFun_RedBoard_Artemis/program .pio/build/SparkFun_RedBoard_Artemis/firmware.bin
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  10.8% (used 42612 bytes from 393216 bytes)
Flash: [==        ]  17.3% (used 170076 bytes from 983040 bytes)
```

With `build_unflags=-O0` and `build_flags=-Os`:
```
Linking .pio/build/SparkFun_RedBoard_Artemis/program
Checking size .pio/build/SparkFun_RedBoard_Artemis/program
arm-none-eabi-objcopy -O binary .pio/build/SparkFun_RedBoard_Artemis/program .pio/build/SparkFun_RedBoard_Artemis/firmware.bin
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  10.8% (used 42568 bytes from 393216 bytes)
Flash: [=         ]  11.4% (used 112228 bytes from 983040 bytes)
```

The output file size for the test with just `build_flags=-Os` and with `build_unflags=-O0` is the same at: 42568 bytes

So to me it looks like there is no real issue, what do you think?

fbenamrouche, 2024-01-22:
Hi @nigelb 

Sorry for the delay. 
I discovered the unflag option, so indeed it works like that.
For for me we can say that it's not a real issue ;)

Thanks

closed 2024-01-22 by fbenamrouche