nigelb/platform-apollo3blue #54: Missing CXXFLAGS for c++

fbenamrouche, 2025-07-31:
Hello,

I'm trying to build a c++ project. My issue is that the compiler complains about mismatching c++ flags (mainly about vfp)
Looking at the builder, the default flags are in CFLAGS... so C only instead of CCFLAGS as far as I understand
(https://community.platformio.org/t/language-specific-build-flags/34559)

I will try to set them in a build script, should be good enough, so not a blocker but just wanted to let you know!

Thanks

nigelb, 2025-08-02:
Hi @fbenamrouche ,

Which SDK are you using?

The Sparkfun Arduino Framework or the AmbiqSDK?

Nigel.

fbenamrouche, 2025-08-02:
Hi @nigelb ,
I'm using AmbiqSDK!

nigelb, 2025-08-02:
Ah ok, I have a branch I have been working on to fix this. It is called  [ambiq_c++](https://github.com/nigelb/platform-apollo3blue/tree/ambiq_c%2B%2B?tab=readme-ov-file)

It currently has the build flags fixed but  if you want to use things like std::vector it needs an update for the linker script to define some missing variables.

You can configure a custom linker script for your project in your platformio.ini file like [this](https://github.com/nigelb/platform-apollo3blue/blob/master/docs/platform-apollo3blue/PlatformIO_ini_Options.md#custom-linker-script)

I will fix this up shortly, now that someone is actually trying to use it 😂

fbenamrouche, 2025-08-03:
yes, this is what I'm going for now. That's why I'm saying that it's not a blocker for me ;)

Thanks!